ros-infrastructure/rosdep #637: Solus support

babba22, 2018-10-29:
I was trying to install ros in [solus](https://getsol.us/home/), but when I have to run `rosdep update` I get this error:

```
rosdep version: 0.12.2

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 386, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 395, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 592, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python3.6/site-packages/rosdep2/sources_list.py", line 490, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3.6/site-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python3.6/site-packages/rosdep2/__init__.py", line 87, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python3.6/site-packages/rosdep2/platforms/debian.py", line 53, in register_platforms
    register_linaro(context)
  File "/usr/lib/python3.6/site-packages/rosdep2/platforms/debian.py", line 70, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python3.6/site-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python3.6/site-packages/rospkg/os_detect.py", line 660, in get_name
    self.detect_os()
  File "/usr/lib/python3.6/site-packages/rospkg/os_detect.py", line 630, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'slackware', 'rhel', 'qnx', 'osx', 'tizen', 'opensuse', 'opensuse', 'opensuse', 'neon', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'centos', 'manjaro', 'arch', 'alpine']
```

Reporting this here as written in the output of the command.

tfoote, 2018-10-29:
solus is not an OS we currently have support for. Please see the developer guide for how to contribute os detection for solus: http://docs.ros.org/independent/api/rosdep/html/developers_guide.html

An example of adding an OS detector is here: https://github.com/ros-infrastructure/rospkg/pull/85

Solus will also need support for a new package manager/platform added in rosdep https://github.com/ros-infrastructure/rosdep/tree/master/src/rosdep2/platforms

Contributions for supporting solus would be appreciated. However I'm going to close this as we're not going to track adding that support.

closed 2018-10-29 by tfoote

babba22, 2018-10-29:
ok, thank you for your explanation!
I will give it a look and see if I am able to contribute :+1: