ros-infrastructure/rosdep #649: Locally ignore a rosdep keys using empty list of packages

ruffsl, 2019-01-17:
From reading the help prompt for `rosdep install`, it is unclear how forme an `empty list of packages` to more permanently ignore rosdep keys by creating a local rosdep rule, i.e.  https://github.com/ros-infrastructure/rosdep/commit/e6afc9f53adfc72cc83d8e438a1ae972e2f7e789

https://github.com/ros-infrastructure/rosdep/blob/558e1720577dd47454e4008d5893b8fccd518802/src/rosdep2/main.py#L300-L309

I've created an `empty list of packages` like so:

``` yaml
console_bridge: {}
fastcdr: {}
fastrtps: {}
libopensplice67: {}
libopensplice69: {}
osrf_testing_tools_cpp: {}
poco_vendor: {}
rmw_connext_cpp: {}
rosidl_typesupport_connext_c: {}
rosidl_typesupport_connext_cpp: {}
rti-connext-dds-5.3.1: {}
tinyxml_vendor: {}
tinyxml2_vendor: {}
urdfdom: {}
urdfdom_headers: {}
```

I'm attempting to use this method suggested above to permanently ignore the rosdep keys in the `core.yaml` file above without having to include them in the `--skip-keys` for each install command call. Something like so: 

``` Dockerfile
COPY core.yaml /etc/ros/rosdep/
RUN echo "yaml file:///etc/ros/rosdep/core.yaml" > /etc/ros/rosdep/sources.list.d/5-default.list

# install ros2 fat archives
...

# install dependencies
RUN apt-get update && rosdep update && rosdep install -y \
    --from-paths /opt/ros/$ROS_DISTRO/share \
    --ignore-src \
    && rm -rf /var/lib/apt/lists/*
```

Related:
* https://index.ros.org/doc/ros2/Installation/Linux-Install-Binary/#installing-the-missing-dependencies
* https://github.com/ros-infrastructure/rosdep/issues/375

NikolausDemmel, 2019-01-17:
This should work I think:

```
console_bridge: {ubuntu: []}
fastcdr: {ubuntu: []}
...
```

Explanation:

`rosdep` searches the sources for the first rule that matches the current operating system and uses that. In your example you provide the keys, but no actual rules, so it is just ignored.

`ubuntu: []` is a shorthand rule that matches all versions of ubuntu and exands to an empty list of apt packages.

ruffsl, 2019-01-18:
Thanks @NikolausDemmel !

closed 2019-01-18 by ruffsl