ros-infrastructure/rosdep #653: Support for format three package.xml

AAlon, 2019-01-23:
Hey,
Looking at REP-149 http://www.ros.org/reps/rep-0149.html it looks like the support should already exist as the status is "Final". However, it seems like the tool does not recognize the `condition` option. Or maybe I'm just defining it wrong. 
The motivation is being able to use catkin's CMake test macros when building for ROS1 and ament's macros when building for ROS2. Other than that the code itself is ROS-agnostic.
#### Configuration
package.xml has:
```
...
<package format="3">
...
<build_depend condition="$ROS_VERSION == 1">catkin</build_depend>
<build_depend condition="$ROS_VERSION == 2">ament_cmake_gtest</build_depend>
```
#### Error encountered
When using rosdep under ROS2 I'm getting `Cannot locate rosdep definition for [catkin]`
When using rosdep under ROS1 I'm getting `Cannot locate rosdep definition for [ament_cmake_gtest]`

#### Full log:
```
$ rosdep --version
0.14.0
```
```
$ echo $ROS_DISTRO
kinetic
$ echo $ROS_VERSION
1
$ rosdep install --from-paths src/utils-common --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
aws_common: Cannot locate rosdep definition for [ament_cmake_gtest]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

```
$ echo $ROS_DISTRO
crystal
$ echo $ROS_VERSION
2
$ rosdep install --from-paths src/utils-common --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
aws_common: Cannot locate rosdep definition for [catkin]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
I looked briefly but didn't see any portion of the code/docs explicitly calling out rep149 support, so I'd appreciate your help figuring out if this is really an issue in rosdep or just misconfiguration on my end.
Thanks!

dirk-thomas, 2019-01-23:
It seems that this wasn't implemented in `rosdep`. Please try the patch from #654 and comment if that works for you.

AAlon, 2019-01-24:
Yep, #654 fixed it :) Thanks for the quick fix!

closed 2019-01-24 by nuclearsandwich

JWhitleyAStuff, 2019-03-25:
@dirk-thomas - This PR fixed the issue in rosdep. However, it looks like the official XSD (http://download.ros.org/schema/package_format3.xsd) hasn't been updated to support the condition attribute. As such, `xmllint` fails when attempting to validate package.xml files which use it. Where can I request a change to the XSD?

dirk-thomas, 2019-03-25:
The `xsd` files are stored in the [REP](https://github.com/ros-infrastructure/rep/) repository. An issue there (or even better a pull request updating the schema file) would be best.

PR #654 (2019-01-24): evaluate conditions before collecting dependencies
Fixes #653.