ros-infrastructure/rosdep #664: rosdep update failing in Ubuntu 16.04, 18.04, Mac

subvertallchris, 2019-02-19:
Trying to run `rosdep update` in multiple environments -- not only different boxes but also different physical networks -- I'm running into the same error which leads me to believe this is a bug somewhere.

```
$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
	Failed to download target platform data for gbpdistro:
	HTTP Error 503: Backend unavailable, connection timeout
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Add distro "bouncy"
Add distro "crystal"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Add distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Add distro "lunar"
Add distro "melodic"
updated cache in /home/bbi/.ros/rosdep/sources.cache
ERROR: Not all sources were able to be updated.
[[[
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
	Failed to download target platform data for gbpdistro:
	HTTP Error 503: Backend unavailable, connection timeout
]]]
```

I get that same error in the following environments:

* Ubuntu 16.04 VM
* On my cloud CI service, building from Docker using an 18.04 image that previous worked
* Trying to build the same image on my Mac

I'm able to `curl https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml` without issue. If I had to guess, it's a dependency listed in that yaml file that's failing but I haven't checked each one. Since I'm using melodic, I'm guessing I can comment out the fuerte line in `/etc/ros/rosdep/sources.list.d/20-default.list` but I want to bring this up since it does not at all look like a problem on my end.

sloretz, 2019-02-19:
fixed by ros-infrastructure/rosdep#663

subvertallchris, 2019-02-19:
Definitely the fastest response I've ever had to an issue on Github.

wjwwood, 2019-02-20:
Should be fixed now, see: https://github.com/ros-infrastructure/rosdep/pull/663#issuecomment-465362397

closed 2019-02-20 by wjwwood