ros-infrastructure/rosdep #666: rosdep requires curl to work, but does not come with it on bionic

sloretz, 2019-02-26:
1. Start a fresh bionic docker container
   ```
   docker run -it ubuntu:bionic bash
   ```
1. install rosdep
   ```
   apt update && apt install -y python-rosdep2
   ```
1. initialize rosdep, it fails
   ```
   # rosdep init
   ERROR: cannot download default sources list from:
   https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
   Website may be down.
   ```
1. Install curl
   ```
   apt install -y curl
   ```
1. Initialize rosdep again, this time it works
   ```
   # rosdep init
   Wrote /etc/ros/rosdep/sources.list.d/20-default.list
   Recommended: please run

   	rosdep update
   ```

I would have expected `curl` to have been installed automatically when `python-rosdep2` was installed.

tfoote, 2019-02-26:
It doesn't actually need curl, but in the stripped down docker instance it needs `ca-certificates` which is in most base Ubuntu images., but no the stripped down docker ones.

sloretz, 2019-02-26:
@tfoote I updated the issue and double checked that installing `ca-certificates` is enough to fix it.

sloretz, 2019-02-26:
Ah I guess this is not a bug for this repo. `stdeb.cfg` has `ca-certificates` already, and it looks like this was reported and fixed 3 years ago in 454.

https://github.com/ros-infrastructure/rosdep/blob/fe63aaac265a2ef3823726250815c0d64e9e9b5c/stdeb.cfg#L2-L3

The package I'm using is not maintained by us.

```
# apt show python-rosdep2
Package: python-rosdep2
Version: 0.11.8-1
Priority: optional
Section: universe/python
Source: ros-rosdep
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 287 kB
Provides: python-rosdep
Depends: python-catkin-pkg, python-rosdistro, python-rospkg, python-yaml, python:any (<< 2.8), python:any (>= 2.7.5-5~)
```

closed 2019-02-26 by sloretz

sloretz, 2019-02-26:
Filed bug on launchpad https://bugs.launchpad.net/ubuntu/+source/ros-rosdep/+bug/1817773