ros-infrastructure/rosdep #671: Accelerate APT installs by executing apt-get once

stertingen, 2019-03-26:
Right now rosdep executes apt-get once for every unmet dependency. Since these dependencies may depend on each other, some package installed by rosdep in a later step may be installed as dependency for an earlier package. Rosdep does not know that and tries to install that package again, apt-get does nothing but consumes time.

To accelerate some automated builds, I use `apt-get install -y $(rosdep check -i --from-paths /tmp | sed -n 's/^apt\s\+//p' | tr '\n' ' ')` instead of `rosdep install -i -y --from-paths /tmp`, since it calls apt-get once which eliminates duplicate depedencies.

I would like to see this as a default behavior: Execute the install command (if the package manager supports it) once with a list of all packages instead once of for each package.

Alternatively, a script-friendly output for dumping all packages to stdout eliminating the use of sed (and tr) in the above command would satisfy the issue.

wjwwood, 2019-03-26:
There is a case you need to consider which is where something needs to be installed from apt, then from something else, like pip, and then yet another things from apt again. Specifically for apt, this is not a practical issue since apt never depends on other package managers, but this case exists (or existed in the past for Homebrew).

Secondly, I think there's some value in having repeatable, deterministic behavior from rosdep. If you coalesced apt installs to a single command, then it might be harder to determine which dependency failed to install in apt (it is more clear when you install them one at a time), and also if you try to install many things, some install and some fail, then you run rosdep again to reproduce you'll end up invoking a different command to apt, again making troubleshooting more difficult.

----

So, I think a script friendly output would be good to have, as long as it accounted for ordering between package manager calls.

And I think an option to have rosdep automatically coalesce apt calls into fewer apt calls would also be ok, though I'm not sure about it being the default.

tfoote, 2019-03-27:
It used to be that rosdep would actually just output a script. And we moved to invoking the installers specifically per package. This was when we also added support for checking that something's installed already. 

One of the big reasons for separating it out was that  debugging failures to install is almost impossible if the packages are aggregated. If rosdep determines that there needs to be 100 packages to install but only one errors it will just say that the request is uninstallable and quit. It doesn't give helpful diagnostics. This can lead to one bad rosdep rule or package resolved by rosdep to make the default ROS installation grind to a halt even if it's a rosdep for a leaf package.

With respect to the speed up I don't know how often the stated overlapping dependencies happen. That could also be worked around by checking again at intermediate states for whether the package is already installed. It depends on how often that occurs as to whether it would actually be faster as it would cause more slow down on average to optimize for the overlapping case. 

Some basic timings show that a no-op apt call on my system is well less than a second. Although this is technically unnecessary work I would put this at a low priority for optimization.

Timings actual install vs no-op
```
$ time sudo apt-get install ros-kinetic-roscpp
[sudo] password for tfoote: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be upgraded:
  ros-kinetic-roscpp
1 upgraded, 0 newly installed, 0 to remove and 277 not upgraded.
Need to get 500 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-roscpp amd64 1.12.14-0xenial-20190320-122631-0800 [500 kB]
Fetched 500 kB in 0s (591 kB/s)            
(Reading database ... 557644 files and directories currently installed.)
Preparing to unpack .../ros-kinetic-roscpp_1.12.14-0xenial-20190320-122631-0800_amd64.deb ...
Unpacking ros-kinetic-roscpp (1.12.14-0xenial-20190320-122631-0800) over (1.12.14-0xenial-20180824-091453-0800) ...
Setting up ros-kinetic-roscpp (1.12.14-0xenial-20190320-122631-0800) ...

real	0m10.227s
user	0m3.188s
sys	0m0.457s
tfoote@snowman:~ Last: [0] (10s Seconds)
$ time sudo apt-get install ros-kinetic-roscpp
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-kinetic-roscpp is already the newest version (1.12.14-0xenial-20190320-122631-0800).
ros-kinetic-roscpp set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 277 not upgraded.

real	0m0.768s
user	0m0.723s
sys	0m0.039s
```

rotu, 2020-04-02:
It would be nice to at least call `apt-get` only once for grouped dependencies like libpcl-all, where the multiple invocations of apt make for messy output.

invoking apt-get directly:
```
sudo apt-get install libpcl-apps1.10 libpcl-common1.10 libpcl-features1.10 libpcl-filters1.10 libpcl-io1.10 libpcl-kdtree1.10 libpcl-keypoints1.10 libpcl-ml1.10 libpcl-octree1.10 libpcl-outofcore1.10 libpcl-people1.10 libpcl-recognition1.10 libpcl-registration1.10 libpcl-sample-consensus1.10 libpcl-search1.10 libpcl-segmentation1.10 libpcl-stereo1.10 libpcl-surface1.10 libpcl-tracking1.10 libpcl-visualization1.10
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cython3 libflann-dev libflann1.9 libfontconfig1-dev libgl2ps-dev libglew-dev libhdf5-mpi-dev libhdf5-openmpi-dev liblz4-dev libnetcdf-cxx-legacy-dev libogg-dev libopenni-dev libopenni2-dev libqt5designercomponents5
  libqt5positioning5 libqt5quickparticles5 libqt5quickshapes5 libqt5quicktest5 libqt5quickwidgets5 libqt5sensors5 libqt5webchannel5 libqt5webkit5 libqt5webkit5-dev libqt5x11extras5 libtheora-dev libvtk7-dev libvtk7-java
  libvtk7-jni libvtk7-qt-dev libvtk7.1p-qt libxft-dev libxss-dev python3-autobahn python3-automat python3-cbor python3-click python3-colorama python3-constantly python3-hamcrest python3-hyperlink python3-incremental
  python3-lz4 python3-mpi4py python3-png python3-pyasn1 python3-pyasn1-modules python3-pyqrcode python3-service-identity python3-snappy python3-trie python3-twisted python3-twisted-bin python3-txaio python3-u-msgpack
  python3-ubjson python3-vtk7 python3-wsaccel python3-zope.interface qdoc-qt5 qhelpgenerator-qt5 qt5-assistant qtattributionsscanner-qt5 qtdeclarative5-dev qtdeclarative5-dev-tools qttools5-dev qttools5-dev-tools
  qttools5-private-dev tcl-dev tcl-vtk7 tcl8.6-dev tk-dev tk8.6-dev vtk7 x11proto-scrnsaver-dev
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libpcl-apps1.10 libpcl-common1.10 libpcl-features1.10 libpcl-filters1.10 libpcl-io1.10 libpcl-kdtree1.10 libpcl-keypoints1.10 libpcl-ml1.10 libpcl-octree1.10 libpcl-outofcore1.10 libpcl-people1.10 libpcl-recognition1.10
  libpcl-registration1.10 libpcl-sample-consensus1.10 libpcl-search1.10 libpcl-segmentation1.10 libpcl-stereo1.10 libpcl-surface1.10 libpcl-tracking1.10 libpcl-visualization1.10
0 upgraded, 20 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/12.6 MB of archives.
After this operation, 119 MB of additional disk space will be used.
Selecting previously unselected package libpcl-common1.10:amd64.
(Reading database ... 304339 files and directories currently installed.)
Preparing to unpack .../00-libpcl-common1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-common1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-kdtree1.10:amd64.
Preparing to unpack .../01-libpcl-kdtree1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-kdtree1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-octree1.10:amd64.
Preparing to unpack .../02-libpcl-octree1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-octree1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-sample-consensus1.10:amd64.
Preparing to unpack .../03-libpcl-sample-consensus1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-sample-consensus1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-search1.10:amd64.
Preparing to unpack .../04-libpcl-search1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-search1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-filters1.10:amd64.
Preparing to unpack .../05-libpcl-filters1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-filters1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-features1.10:amd64.
Preparing to unpack .../06-libpcl-features1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-features1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-ml1.10:amd64.
Preparing to unpack .../07-libpcl-ml1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-ml1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-segmentation1.10:amd64.
Preparing to unpack .../08-libpcl-segmentation1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-segmentation1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-surface1.10:amd64.
Preparing to unpack .../09-libpcl-surface1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-surface1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-apps1.10:amd64.
Preparing to unpack .../10-libpcl-apps1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-apps1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-io1.10:amd64.
Preparing to unpack .../11-libpcl-io1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-io1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-keypoints1.10:amd64.
Preparing to unpack .../12-libpcl-keypoints1.10_1.10.0+dfsg-5ubuntu1_amd64.deb ...
Unpacking libpcl-keypoints1.10:amd64 (1.10.0+dfsg-5ubuntu1) ...
Selecting previously unselected package libpcl-outofcore1.10:amd64.
[... cut ...]
  python3-constantly python3-hamcrest python3-hyperlink python3-incremental python3-lz4 python3-mpi4py python3-png python3-pyasn1 python3-pyasn1-modules python3-pyqrcode python3-service-identity python3-snappy python3-trie
  python3-twisted python3-twisted-bin python3-txaio python3-u-msgpack python3-ubjson python3-vtk7 python3-wsaccel python3-zope.interface qdoc-qt5 qhelpgenerator-qt5 qt5-assistant qtattributionsscanner-qt5 qtdeclarative5-dev
  qtdeclarative5-dev-tools qttools5-dev qttools5-dev-tools qttools5-private-dev tcl-dev tcl-vtk7 tcl8.6-dev tk-dev tk8.6-dev vtk7 x11proto-scrnsaver-dev
Use 'sudo apt autoremove' to remove them.
...
```