ros-infrastructure/rosdep #672: MX Linux not "supported"

gstavrinos, 2019-03-30:
I think adding support for MX Linux would help quite a lot of people, with MX Linux gaining in popularity everyday. It is Debian based, so it should not be a problem at all. I  am currently running MX Linux 18.1 with ROS Melodic, with no problems.

Thanks,

George

tfoote, 2019-04-01:
We'd be happy to accept a pull request to add support. 

Note that you'll want to start with an OsDetector in rospkg https://github.com/ros-infrastructure/rospkg/blob/master/src/rospkg/os_detect.py

http://docs.ros.org/independent/api/rospkg/html/developers_guide.html#developing-new-osdetectors

Then you can add an alias entry if it aliases to an ubuntu or debian distribution like here for elementary https://github.com/ros-infrastructure/rosdep/blob/fe63aaac265a2ef3823726250815c0d64e9e9b5c/src/rosdep2/platforms/debian.py#L76

gstavrinos, 2019-04-02:
Thanks for the guidelines. I will work on it asap.

gstavrinos, 2019-04-02:
Made the required PRs #674 and ros-infrastructure/rospkg#167.

closed 2019-04-02 by gstavrinos