ros-infrastructure/rosdep #677: Python3 test failures

ross-desmond, 2019-05-03:
Failures have been occurring in the Python3 CI for approximately 2 months:
https://travis-ci.org/ros-infrastructure/rosdep/builds/506177460

tfoote, 2019-05-03:
This is dueo to new warnings since yaml.load has had strong warnings added in python3. There's a fix for the elements we triaged as critical in #651 before it was escalated in python3 and c7a76ef in #675 has fixes for the remaining instances.

ross-desmond, 2019-05-03:
The reason I created an issue for this is we are seeing ssl errors occurring recently (within the last 3 days) in our CI builds (no changes to these repos). Have you seen this issue within Travis with rosdep recently?
```
ERROR: error loading sources list:
	<urlopen error <urlopen error ('_ssl.c:574: The handshake operation timed out',)> (https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml)>
```

LanderU, 2019-05-04:
We're currently having the same issue in our CI.

ipa-rmb, 2019-05-06:
Same issues here.

tfoote, 2019-05-07:
ssl errors like that are more likely related to the connection between you and github rather than a code issue. We're not seeing any issues like that on the CI here nor as far as I know on our buildfarm which uses rosdep heavily. 

There's [one test failing now on master](https://travis-ci.org/ros-infrastructure/rosdep/jobs/389515351) related to linters on python3.6 only. I think that platform likely got a new version of flake8 though I'm surprised that all the PR tests prior to testing on master passed. There must be a slightly different environment.

tfoote, 2019-05-07:
CI is clean after #678 was merged

closed 2019-05-07 by tfoote