ros-infrastructure/rosdep #681: Can't intall python dependencies in ROS 2

crdelsey, 2019-05-13:
This is related to #519 and #542 

I get the following error when trying to install a python dependency using rosdep in a `Dashing` release.

```
ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep
rosdep version: 0.15.1
Bad installer [pip]: [Errno 2] No such file or directory: 'pip': 'pip'
```

Since the default ROS 2 installation uses python3, we need `rosdep` to install the packages using `pip3`, however there is no way to specify that.

You can see a failing build log here:
https://travis-ci.org/ros-planning/navigation2/builds/531898111?utm_source=github_status&utm_medium=notification

russkel, 2019-05-31:
I discovered that using pipenv can alleviate some of these issues. It symlinks a `pip` and isolates the packages from the system.

sloretz, 2019-11-12:
I think this is fixed by #709. Would you mind checking using rosdep >=0.17.0?

crdelsey, 2019-11-12:
I can no longer reproduce the problem. Looks like it is fixed. Thanks!

closed 2019-11-12 by crdelsey