ros-infrastructure/rosdep #682: Prefer pip for python packages

russkel, 2019-05-25:
Is there a way for rosdep to use the pip package over the distro's pkg for python dependencies?

If not, if I create a patch that adds a `--prefer-pip` or a general `--prefer=pip` option would this be accepted?

tfoote, 2019-05-29:
I don't think that this is necessarily how I would recommend approaching this problem. The goal of the rosdep database is that for a given distribution and version you will know exactly what you're getting and it will be repeatable for all developers. And this should also be consistent over time. This is why we strongly prefer the binaries packaged with distributions as they are reproducible across installations. pip rules are also not able to be packaged.

I don't know your use case exactly, but if you want to prefer pip over the system packages my recommendation would be to create your own rosdep.yaml file and prepend it to your sources.list. Then for any installation you're doing you have full control of the versions. And you and your users will know that they're using the modified version of the keys that come from your rosdep sources. 

Otherwise the version in the default rosdistro becomes unreliable, and depends on what options someone else might have run at some point in the past on that same machine.

russkel, 2019-05-29:
> The goal of the rosdep database is that for a given distribution and version you will know exactly what you're getting and it will be repeatable for all developers.

> And this should also be consistent over time.

Are we getting this with this implementation though? I would bet the packaged versions of the python dependencies range from distro to distro. At least with pip we could force a particular version (and actually have it available through PyPI). 

> pip rules are also not able to be packaged.

Could this support be added?

> I don't know your use case exactly,

I'd prefer to get the packages directly from the source rather than the -in my experience- unreliable, out of date and inflexible maintained packages. With pip installed packages I actually know what's going on and I know how to update/remove/reinstall them.

I'm not really interested in changing people's opinion about system packaged python packages, but if I'm going to work in support for pip first, I would at least like to know if/how it would be accepted as a contribution.

tfoote, 2019-05-29:
>> The goal of the rosdep database is that for a given distribution and version you will know exactly what you're getting and it will be repeatable for all developers.
>> 
>> And this should also be consistent over time.

>
> Are we getting this with this implementation though? I would bet the packaged versions of the python dependencies range from distro to distro. At least with pip we could force a particular version (and actually have it available through PyPI).

When I say consistent in time I'm talking about being the same on the same platform. The version in bionic will be very similar to the version in bionic at  the end. Even if upstream publishes a new full version number higher that won't be backported to bionic to keep the experience consistent.

>> pip rules are also not able to be packaged.

> Could this support be added?

Generally no. I don't know of any native package manager that supports dependencies on other package managers. rosdep does have basic support for declaring a dependency on a package from another manager. But it's a meta-package manager that delegates to package managers. 

There are people who are making a great effort to help package and maintain many of these package on the debian platforms for all of us to benefit. And in another approach there are people packaging some pure python libraries into ROS packages for easier distribution. 

> I'd prefer to get the packages directly from the source rather than the -in my experience- unreliable, out of date and inflexible maintained packages. With pip installed packages I actually know what's going on and I know how to update/remove/reinstall them.

Unfortunately this is exactly the reasons that we generally use the maintianed packages. You can turn that description around and refer to them as stable and tested. Indeed it's much easier to update via pip, but that's actually where it will bite back. One of the really valuable things that the distribution packaged versions of packages provides is that it has been tested and qa'd with all the other packages available on your system. Any time you change a library dependency you should be checking that it works with all the downstream packages that depend on it. 

I think you'll find that removing and reinstalling debian packages is actually much more robust than pip packages. pip has [documented](https://pip.pypa.io/en/stable/reference/pip_uninstall/) types of packages it cannot uninstall. 

As I'm trying to suggest I don't think that having a --prefer-piip option is a good idea. As you've mentioned you would likely need to call out the specific version of every pip package that you're creating rules for and you'll likely only be testing on a specific platform or two. Thus all the existing extensibility should be able to resolve your use case by maintaining the set of rules that prioritizes pip resolutions of the dependencies. This will let you pick your level of tolderance for breakage and updates and not end up trying to embed this conditional logic in the upstream database

What you want to do is maintain a parallel database with a different policy for priority. If users want to follow that different priority they can add your database to their sources. What we don't want is a standard user to accidentally end up with content from mixed policies with the expectation that everything will just work.

If we were to start adding these packages to the main database we would also be implicitly telling ROS package mantainers that this is a new configuration that they are also expected to have validated and tested against which is also not something that we want to do either. We very explicitly scope what configurations (os, distro, arch)  should be tested for each rosdistro and this would add another dimension to try to get coverage.

So overall I guess what I'm saying is that I'd strongly suggest simply building up your own set of rules that you can validate and test instead of trying to add an option to rosdep and ask us to intermingle your alternative policy keys with the main set of rules.

russkel, 2019-05-29:
> Thus all the existing extensibility should be able to resolve your use case by maintaining the set of rules that prioritizes pip resolutions of the dependencies. This will let you pick your level of [...]

This sounds pretty good, I'll try this way and see how it works out. Thanks.

closed 2019-05-29 by russkel