ros-infrastructure/rosdep #683: rosdep experienced an error

jidibinlin, 2019-05-26:


jidibinlin, 2019-05-26:
I used the source way to install the ros and when I execute "rosdep install --from-paths src --ignore-src --rosdistro melodic -y"  this happend

ERROR: Rosdep experienced an error: rosdep OS definition for [python-opencv:fedora] must be a dictionary, string, or list: None
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.15.0

Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 396, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 492, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 691, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python3.7/site-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python3.7/site-packages/rosdep2/lookup.py", line 401, in resolve_all
    self.resolve(rosdep_key, resource_name, installer_context)
  File "/usr/lib/python3.7/site-packages/rosdep2/lookup.py", line 485, in resolve
    installer_key, rosdep_args_dict = definition.get_rule_for_platform(os_name, os_version, installer_keys, default_key)
  File "/usr/lib/python3.7/site-packages/rosdep2/lookup.py", line 148, in get_rule_for_platform
    raise InvalidData('rosdep OS definition for [%s:%s] must be a dictionary, string, or list: %s' % (self.rosdep_key, os_name, data), origin=self.origin)
rosdep2.core.InvalidData: rosdep OS definition for [python-opencv:fedora] must be a dictionary, string, or list: None

tfoote, 2019-05-29:
It looks like only fedora 29 has a resolution for that package: https://github.com/ros/rosdistro/blob/5fa0f5d44fb661342c05b5cfac8ee3f891dfb154/rosdep/python.yaml#L2522 if you're on a different version of fedora it looks like they dropped support for 30 and newer: https://github.com/ros/rosdistro/pull/21325 and fedora 28 just went out of support

closed 2019-05-29 by tfoote

lukehutch, 2019-07-10:
I ran into this too... rosdep really needs to provide a more useful error message if a package is not found.