ros-infrastructure/rosdep #689: Rosdep experienced an error: rosdep OS definition for [python-opencv:fedora] must be a dictionary, string, or list: None

lukehutch, 2019-07-10:
Following the instructions on http://wiki.ros.org/Installation/Source on Fedora 30, I get:

```
$ rosdep install --from-paths src --ignore-src --rosdistro melodic -y
bash: rosdep: command not found...
Install package 'python3-rosdep' to provide command 'rosdep'? [N/y] y

 * Waiting in queue... 
 * Loading list of packages.... 
 * Downloading details about the software repositories.... 
 * Loading list of packages.... 
 * Downloading details about the software repositories.... 
 * Loading list of packages.... 
The following packages have to be installed:
 python3-rosdep-0.15.2-1.fc30.noarch	HTML documentation for 'python-rosdep'
Proceed with changes? [N/y] y

 * Waiting in queue... 
 * Waiting for authentication... 
 * Waiting in queue... 
 * Downloading packages... 
 * Requesting data... 
 * Testing changes... 
 * Installing packages... 

ERROR: Rosdep experienced an error: rosdep OS definition for [python-opencv:fedora] must be a dictionary, string, or list: None
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.15.2

Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 396, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 492, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python3.7/site-packages/rosdep2/main.py", line 691, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python3.7/site-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python3.7/site-packages/rosdep2/lookup.py", line 401, in resolve_all
    self.resolve(rosdep_key, resource_name, installer_context)
  File "/usr/lib/python3.7/site-packages/rosdep2/lookup.py", line 485, in resolve
    installer_key, rosdep_args_dict = definition.get_rule_for_platform(os_name, os_version, installer_keys, default_key)
  File "/usr/lib/python3.7/site-packages/rosdep2/lookup.py", line 148, in get_rule_for_platform
    raise InvalidData('rosdep OS definition for [%s:%s] must be a dictionary, string, or list: %s' % (self.rosdep_key, os_name, data), origin=self.origin)
rosdep2.core.InvalidData: rosdep OS definition for [python-opencv:fedora] must be a dictionary, string, or list: None
```

lukehutch, 2019-07-10:
The error is the same in both Python 2 and Python 3.

lukehutch, 2019-07-10:
I found this bug which explains the problem... https://github.com/ros-infrastructure/rosdep/issues/683

exoticDFT, 2019-07-22:
I am also using Fedora 30 and receiving the same issue. Looking at issue #683 it seems the solution was to just remove support for fedora outside of fedora 29. Is that correct? I have opencv installed for python2, through pip installation `pip2 install --user opencv-python`.

Is there a way around this error during the ROS build process, can we just not install ROS on Fedora anymore or am I missing something with the resolution of #683?

cottsay, 2019-07-22:
You can still invoke rosdep if you skip the python2 key. Add `--skip-keys opencv-python` to your rosdep command.

exoticDFT, 2019-07-22:
@cottsay Thank you for the reply. The addition of `--skip-keys=python-opencv` worked to get me past this issue.