ros-infrastructure/rosdep #698: Rosdep can't find ruamel.yaml package

rickstaa, 2019-08-13:
When I try to specify the `ruamel.yaml` package as a ROS package dependency in my `package.xml` file I get the following error:

```
rosdep install tutorial_package
executing command [sudo -H apt-get install python-ruamel.yaml]

ERROR: Rosdep experienced an error: [Errno 2] No such file or directory
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.15.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 396, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 492, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 715, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 525, in install
    verbose=verbose, quiet=quiet)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 597, in install_resolved
    result = run_command(sub_command, installer_key, failures, verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 578, in run_command
    result = subprocess.call(command)
  File "/usr/lib/python2.7/subprocess.py", line 523, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

## Package.xml
```
<exec_depend>python-ruamel.yaml</exec_depend>
```

## rosdep command
```
rosdep install --from-paths src --ignore-src -r -y
```

tfoote, 2019-08-16:
What is your environment? It's not clear what file or directory is missing, but my guess is that your platform doesn't have `sudo` installed.

rickstaa, 2019-08-18:
@tfoote You were a 100% right I'm working inside a singularity container. The problem was that I did not run the container as the root user. Thanks a lot for the help.

closed 2019-08-18 by rickstaa