ros-infrastructure/rosdep #707: Rosdep experienced an error: cannot import name OS_NIXOS

kahkeng, 2019-09-23:
It seems the latest version of rosdep (0.16.1) from debian apt has an issue running certain commands like `update` or `--all-versions`.

I have an older version (0.15.2) in a docker image and these commands seem to work on that version, but fail after I upgrade to the latest. Please see debug output below.

```
root@b24ae7abfd60:/# uname -a
Linux b24ae7abfd60 4.4.0-87-generic #110-Ubuntu SMP Tue Jul 18 12:55:35 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

root@b24ae7abfd60:/# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial

root@ff305f8d6ba1:/# dpkg -l | grep rosdep
ii  python-rosdep                                0.15.2-1                                              all          rosdep package manager abstraction tool for ROS

root@ff305f8d6ba1:/# rosdep --all-versions
0.15.2

Versions of installers:
  apt-get 1.2.32
  pip 19.2.3
  setuptools 41.2.0
  gem 2.5.2.1

root@b24ae7abfd60:/# rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Add distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
updated cache in /root/.ros/rosdep/sources.cache
```

The above start to fail once I upgrade:
```
root@ff305f8d6ba1:/# apt-get update && apt-get install --no-install-recommends -y python-rosdep
...

root@ff305f8d6ba1:/# dpkg -l | grep rosdep
ii  python-rosdep                                0.16.1-1                                              all          rosdep package manager abstraction tool for ROS

root@ff305f8d6ba1:/# rosdep --all-versions
0.16.1

ERROR: Rosdep experienced an error: cannot import name OS_NIXOS
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.16.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 143, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 349, in _rosdep_main
    installers = create_default_installer_context().installers
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 63, in create_default_installer_context
    from .platforms import nix
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/nix.py", line 31, in <module>
    from rospkg.os_detect import OS_NIXOS
ImportError: cannot import name OS_NIXOS

root@ff305f8d6ba1:/# rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Add distro "crystal"

ERROR: Rosdep experienced an error: cannot import name OS_NIXOS
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.16.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 143, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 397, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 406, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 617, in command_update
    skip_eol_distros=not options.include_eol_distros)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 497, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 63, in create_default_installer_context
    from .platforms import nix
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/nix.py", line 31, in <module>
    from rospkg.os_detect import OS_NIXOS
ImportError: cannot import name OS_NIXOS
```

kahkeng, 2019-09-23:
Ok never mind! I found that I had `rospkg==1.1.9` still installed somewhere (perhaps dependency for another package), removing it via `pip uninstall rospkg` to get the latest one exposed resolved this issue for me.

closed 2019-09-23 by kahkeng

rbed23, 2023-08-16:
thanks - ran the following and things worked

`sudo apt purge python3-rospkg`
`pip uninstall rospkg`
`sudo apt install python3-rospkg`