ros-infrastructure/rosdep #710: ERROR: Rosdep experienced an error: hostname

Chuanzhaowang, 2019-09-30:
ERROR: Rosdep experienced an error: hostname 'raw.githubusercontent.com' doesn't match either of 'default.ssl.fastly.net', 'a.ssl.fastly.net', 'control.fastly.net', 'mirrors.fastly.net', 'purge.fastly.net', 'tools.fastly.net', '*.a.ssl.fastly.net', '*.fastly.com', '*.global.ssl.fastly.net', '*.hosts.fastly.net', 'fastly.com'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.16.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 143, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 397, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 406, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 550, in command_init
    data = download_default_sources_list()
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 332, in download_default_sources_list
    f = urlopen(url, timeout=DOWNLOAD_TIMEOUT)
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 429, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 447, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1241, in https_open
    context=self._context)
  File "/usr/lib/python2.7/urllib2.py", line 1195, in do_open
    h.request(req.get_method(), req.get_selector(), req.data, headers)
  File "/usr/lib/python2.7/httplib.py", line 1073, in request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python2.7/httplib.py", line 1113, in _send_request
    self.endheaders(body)
  File "/usr/lib/python2.7/httplib.py", line 1069, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 907, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 869, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 1294, in connect
    server_hostname=server_hostname)
  File "/usr/lib/python2.7/ssl.py", line 353, in wrap_socket
    _context=self)
  File "/usr/lib/python2.7/ssl.py", line 601, in __init__
    self.do_handshake()
  File "/usr/lib/python2.7/ssl.py", line 838, in do_handshake
    match_hostname(self.getpeercert(), self.server_hostname)
  File "/usr/lib/python2.7/ssl.py", line 272, in match_hostname
    % (hostname, ', '.join(map(repr, dnsnames))))
CertificateError: hostname 'raw.githubusercontent.com' doesn't match either of 'default.ssl.fastly.net', 'a.ssl.fastly.net', 'control.fastly.net', 'mirrors.fastly.net', 'purge.fastly.net', 'tools.fastly.net', '*.a.ssl.fastly.net', '*.fastly.com', '*.global.ssl.fastly.net', '*.hosts.fastly.net', 'fastly.com'

tfoote, 2019-10-03:
This looks like you have a certificate error on your system. Make sure that you can access https urls from your python instance. Often this means that you have not installed the package [`ca-certificates`](https://packages.ubuntu.com/xenial/ca-certificates) or there might be a system configuration or network issue.

CarmChao, 2020-02-27:
I have the same problem, have you resolved it

CarmChao, 2020-02-28:
I found a way, just add "ssl._create_default_https_context = ssl._create_unverified_context" to sources_list.py to skip the certificate.

KainanSu, 2022-04-04:
in china, use` rosdepc` in https://mp.weixin.qq.com/s/VGs8oWdhHH6XsHcx21lN4Q, and you will not trap in many trouble with `rosdep`.

mikechen66, 2023-09-12:
While I want to access to the XSUM dataset, I use the simple code snippet.

```
from datasets import load_dataset
raw_datasets = load_dataset("xsum", split="train")
```

Howerver, it shows the following error.

ConnectionError: Couldn't reach https://raw.githubusercontent.com/EdinburghNLP/XSum/master/XSum-Dataset/XSum-TRAINING-DEV-TEST-SPLIT-90-5-5.json (SSLError(MaxRetryError('HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /EdinburghNLP/XSum/master/XSum-Dataset/XSum-TRAINING-DEV-TEST-SPLIT-90-5-5.json (Caused by SSLError(CertificateError("hostname 'raw.githubusercontent.com' doesn't match either of 'default.ssl.fastly.net', 'fastly.com', '.a.ssl.fastly.net', '.hosts.fastly.net', '.global.ssl.fastly.net', '.fastly.com', 'a.ssl.fastly.net', 'purge.fastly.net', 'mirrors.fastly.net', 'control.fastly.net', 'tools.fastly.net'")))')))

I check both certicicate and ssl as follows.

`$ apt-cache policy ca-certificates`

ca-certificates: Installed: 20230311ubuntu0.20.04.1, Candidate: 20230311ubuntu0.20.04.1

`$ apt-cache policy ssl-cert`

ssl-cert: Installed: 1.0.39 Candidate: 1.0.39

What's wrong with the load_data function with the datasets module? Please indicate how to solve the issue.

mikechen66, 2023-09-12:
> I found a way, just add "ssl._create_default_https_context = ssl._create_unverified_context" to sources_list.py to skip the certificate.

CarmChao: 

Where is sources_list.py? 

For Conda, I install git with the following command.

`conda install -c anaconda git`

I got to know that pip is not applicable to install git. 

I use the following code snippet into the ipynb document, but it is now applicable to load_data() function within the datasets module. 
```

try:
    _create_unverified_https_context = ssl._create_unverified_context
except AttributeError:
    pass
else:
    ssl._create_default_https_context = _create_unverified_https_context

Please help solve the issue. 

Thank a lot in advance

```