ros-infrastructure/rosdep #714: ImportError: cannot import name OS_NIXOS

JonisK, 2019-10-08:
Running rosdep in a docker container in gitlab CI results in the following error. The rosdep call executed successfully one week ago.

I am running Ubuntu 16.04 and ROS Kinetic inside the docker container.

``` bash
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.

ERROR: Rosdep experienced an error: cannot import name OS_NIXOS
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.16.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 143, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 397, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 406, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 617, in command_update
    skip_eol_distros=not options.include_eol_distros)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 497, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 63, in create_default_installer_context
    from .platforms import nix
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/nix.py", line 31, in <module>
    from rospkg.os_detect import OS_NIXOS
ImportError: cannot import name OS_NIXOS
```

nuclearsandwich, 2019-10-08:
What version of rospkg is installed on your system. It's possible that we didn't bump the rospkg dependency when we should have?

JonisK, 2019-10-08:
Hi, thank you for the quick reply! rospkg's version number is `1.1.9` in my environment

nuclearsandwich, 2019-10-10:
#717 should resolve this. You can work around the issue locally by manually updating the installed version of rospkg to the latest version.

JonisK, 2019-10-11:
Updating `rospkg` to `1.1.10` solved the issue for me. Thank you for the quick help!

commit 531f3f7b (2019-10-18): Bump rosdep dependency so that NixOS is guaranteed as an available platform.

Closes #714.
PR #717 (2019-10-18): Bump rosdep dependency so that NixOS is guaranteed as an availab…
Closes #714.
commit 288a6c79 (2019-10-18): Bump rosdep dependency so that NixOS is guaranteed as an availab… (#717)

Closes #714.
closed 2019-10-11 by JonisK