ros-infrastructure/rosdep #720: `rosdep resolve` on version 0.16.2 yields `TypeError`

acarrillo, 2019-10-18:
I installed `rosdep` from PyPI. `rosdep resolve` yields the failing the stack trace below:
```
acarrillo ~/code/farmwise_main/util/farmdep $ rosdep resolve

ERROR: Rosdep experienced an error: argument of type 'NoneType' is not iterable
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.16.2

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 406, in _rosdep_main
    if os.environ['ROS_DISTRO'] in python_versions:
TypeError: argument of type 'NoneType' is not iterable
```

acarrillo, 2019-10-18:
I'm using ROS kinetic on Ubuntu 16.

sloretz, 2019-10-18:
Does the error persist after running `rosdep update`?

acarrillo, 2019-10-18:
@sloretz thanks for checking in, I am unable to `rosdep update` as well:

```
acarrillo ~ $ rosdep update

ERROR: Rosdep experienced an error: argument of type 'NoneType' is not iterable
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.16.2

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 406, in _rosdep_main
    if os.environ['ROS_DISTRO'] in python_versions:
TypeError: argument of type 'NoneType' is not iterable

```

sloretz, 2019-10-18:
Hmm, does it persist after doing `ROS_PYTHON_VERSION=2 rosdep update`?

acarrillo, 2019-10-18:
Nope that fixes it -- if I'm reading the intent of #708, that PR was supposed to set that variable if it doesn't exist ( as is the case on my + my CI machines); is some other ROS setup software supposed to be setting `ROS_PYTHON_VERSION`?

PR #721 (2019-10-19): Add default kwarg to MetaDatabase.get()
I think this fixes #720

It should fix broken `rosdep` CLI when a kinetic workspace is sourced. I suspect the environment is:

* There is an existing `kinetic` install, and it's sourced in the current terminal
* `rosdep update` has been successfully run in the past.
* `ros_environment` on `kinetic` does not set `ROS_PYTHON_VERSION`, but it does set `ROS_DISTRO`

Then I think what happened is:

1. Rosdep saw `ROS_DISTRO`, but no `ROS_PYTHON_VERSION`, so it tried to ask the cache which python version to use.
1. `Metadatabase.get()` had no data about the py [... cut]
closed 2019-10-19 by sloretz