ros-infrastructure/rosdep #722: rosdep update throws ROS_PYTHON_VERSION warning

mikaelarguedas, 2019-10-19:
In a system with nothing sourced (no `ROS_*` env var defined)

```
rosdep update
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
```

This does not crash the CLI but seems like a spurious warning. Does rosdep update need to know / do something differently based on the ROS_PYTHON_VERSION used?
Is it expected to run rosdep update with a `--rosdistro` argument to it ?

sloretz, 2019-10-19:
The warning is intentional, and was added in #708. Some packages use `ROS_PYTHON_VERSION` as a conditional dependency in their package.xml (ex: [genpy](https://github.com/ros/genpy/blob/3cc66e76075b04117976962909e7cf8912b08951/package.xml#L25-L26)). If `ROS_PYTHON_VERSION` is unset both of those evaluate to false.

To avoid that `rosdep` tries to pick a default value. If `--rosdistro` or `ROS_DISTRO` are given then the default comes from [the rosdistro index](https://github.com/ros/rosdistro/blob/ff73f6a8d3f10a79d7f4c7c24cba82277ef89825/index-v4.yaml#L65). If that doesn't work, then it picks the version that was used to run rosdep. In that case the warning is printed because that python version might not be related to the workspace you're trying to install dependencies for.

mikaelarguedas, 2019-10-19:
I understand the purpose of the warning when there is a need to resolve / install dependencies. But I don't understand why it's being raised with `rosdep update` is being called.

Another way to phrase it: how should the [tutorials](http://wiki.ros.org/melodic/Installation/Source) be updated for this warning to not be displayed?

xqms, 2019-10-21:
Currently I get mails from `ros_buildfarm` warning me about unstable doc jobs related to this issue.

Example job: http://build.ros.org/job/Kdoc__rosmon__ubuntu_xenial_amd64/9/consoleFull

The warnings are caused by `rosdep init`

```
Step 22/25 : RUN rosdep init
---> Running in 3af4520a3dcb
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

 	rosdep update
```

and a hidden `rosdep update` later on:

```
03:32:01 WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
```

I guess either the `ros_buildfarm` jobs or `rosdep` should be changed to avoid the warning.

sloretz, 2019-10-21:
@xqms @mikaelarguedas mind checking if #725 eliminates the warning in your use cases?

mikaelarguedas, 2019-10-21:
#725 Fixes the warning for me :+1:

xqms, 2019-10-21:
Not really sure how to test it within `ros_buildfarm` (my guess: hack into the used docker images to replace/update `rosdep`), but from the description #725 should fix it :+1:

closed 2019-10-22 by sloretz

sloretz, 2019-10-22:
Fixed in `0.17.1`

PR #725 (2019-10-22): Setup environment only when necessary
Fixes #722 

This refactors the environment variable logic to happen in all rosdep commands except `init`, `update`, and `fix-permissions`.