ros-infrastructure/rosdep #723: Update the database of only a single ROS distribution

mikaelarguedas, 2019-10-19:
feature request / question: would it be possible to specify to rosdep we want to update the database of a single ROS distribution?

Rationale: Often rosdep is slow to run and the more active ROS (1 and 2) distributions the longer it takes to run. Ignoring the EOL distros helped a great bit but it seems like performace would be even better with a single ROS distribution to process.

maybe it could be done if the rosdistro argument is explicitly used `rosdep update --rosdistro kinetic`, or maybe a different flag `--only-selected-rosdistro`?
Also it would override the eol flag compared to currently even if you specify explicitly an eol distro it doesnt get updated if it's eol
```
rosdep update --rosdistro indigo
...
Skip end-of-life distro "indigo"
...
```

ruffsl, 2019-10-19:
Would it make sense for rosdep to default to the ROS_DISTRO env if the arg is unset, same as the `--rosdistro` arg in `rosdep install`? If both env and arg are unset, then the default behavior could be the same as is today for `rosdep update`.

mikaelarguedas, 2019-10-19:
> Would it make sense for rosdep to default to the ROS_DISTRO env if the arg is unset, same as the --rosdistro arg in rosdep install?

Yeas I was going back and forth on that as it is not uncommon to have a given distro sourced by default but you want to update all the database.
e.g. I have melodic sourced in my bashrc and want to get dependencies to build my dshing workspace. I would usually just do
```
rosdep update
rosdep install --rosdistro dashing ...
```
And in that case I wont have the latest state of dashing packages in the db to perform the install.

So my preference would be to not change the behavior unless an argument is passed.
`--rosdistro` seems a sensible choice but maybe maintainers would prefer introducing a new one for existing behavior to stay unchanged

mikaelarguedas, 2019-11-28:
@nuclearsandwich @wjwwood gentle ping. Do you have any feedback on this?

wjwwood, 2020-01-02:
Makes sense to me. 👍 for a pull request for this feature.

> So my preference would be to not change the behavior unless an argument is passed.

I agree.

closed 2020-01-24 by wjwwood