ros-infrastructure/rosdep #732: Python 3 version of python-can is not installable

felixdivo, 2019-11-27:
What is the recommended way of installing packages explicitly for ROS2/Python 3? In this specific case, it's about `python-can`, which is installed for Python 2 for example in the `osrf/ros:dashing-desktop` docker image, despite there only being Python 3 packages present in the ROS2 workspace.

One option would be to use pip (Q: Would it be sensitive to the Python version in the way rosdep utilizes it?) or another option would be to add a `python3-can` package [here](https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L733), pointing to the `python3-can` apt package.

What should I do? I could not find any documentation on this (Is there any?).

Kind of related to #607.

dirk-thomas, 2019-11-27:
Adding `python3-*` keys in the rosdep database is the way to go (see ros-infrastructure/rep#201).

A package can then reference the Python 2 and/or Python 3 specific rosdep key. If necessary both can be referenced depending on the Python version: see e.g. https://github.com/ros/catkin/blob/86afdf7f1b59af72c84e3b4854134e288dfde97e/package.xml#L22-L23

closed 2019-11-27 by dirk-thomas

felixdivo, 2019-11-27:
Okay, great!