ros-infrastructure/rosdep #733: [ROS2] rosdep usage on Windows

seanyen, 2019-12-12:
I'd like to start a thread to discuss the `rosdep` of Windows support for ROS2. From ROS2 tutorials, `rosdep` seems still to play a standard utility for pulling external dependencies for ROS packages. Here is an [example](https://answers.ros.org/question/339282/rosdep-on-windows/) where people is checking the `rosdep` usage on Windows.

Are we expecting `rosdep` to play the same role on Windows platform?

BTW, there are some pull requests to enable `rosdep` to run on Windows in general. Any feedback would be appreciated. 
* #661
* #656

seanyen, 2020-01-22:
@sloretz @dirk-thomas Any feedback on this?

wmmc88, 2021-09-24:
Any chance rosdep is going to be supported in Windows in the future?

wuxiaohua1011, 2022-02-09:
bump on this

zimmy87, 2022-03-24:
Has there been any update on this issue? I'm trying to follow the guide [here](https://ms-iot.github.io/ROSOnWindows/GettingStarted/UsingROSonWindows.html), but I can't seem to get rosdep working on my local Windows ROS2 galactic installation.