ros-infrastructure/rosdep #740: Fails to find pure cmake package when building with Colcon

Levi-Armstrong, 2020-02-06:
Catkin and Colcon both support building pure cmake packages but when building with colcon rosdep fails to find these. The issues it that when building with catkin pure cmake packages are added to ROS_PACKAGE_PATH but with colcon they are not added to AMENT_PREFIX_PATH so rosdep does not locate these package. I am not sure if this is the best approach to resolve this issue but it looks like everything is the CMAKE_PREFIX_PATH and if you filter based on COLCON_PREFIX_PATH you would get both ament packages and pure cmake package built in your workspace when using the--ignore-src.

wjwwood, 2020-02-06:
It's hard to understand if this is a bug or expected behavior, as the use case isn't clear to me.

It might help if you formulated it as:

- steps to reproduce
- expected behavior
- actual behavior

Ideally with something that can be easily reproduced.

Levi-Armstrong, 2020-02-07:
Sure. We ran into an issue when using industrial_ci with colcon to build a repository that had packages in the rosinstall that were pure cmake but included a package.xml file. Here is an example [repository](https://github.com/Levi-Armstrong/ROSDepISSUE). I now believe it may be an issue with industrial_ci because I created two workspace an upstream to contain those package in the rosinstall and one for the repository in the provided link. I first built the upstream workspace, then sourced this upstream workspace and used the the following command `rosdep install -q --from-paths "path_to_provided_workspace_src_directory" --ignore-src` and everything does work. Though industrial_ci fails to ignore package (opw_kinematics) in the upstream, so it must be doing something different. I original thought it was a possible issue with rosdep based on the description below because when using colcon to build pure cmake packages they do not show up in AMENT_PREFIX_PATH. I will also create an issue on industrial_ci.

>'If specified then rosdep will ignore keys that '
                           'are found to be catkin or ament packages anywhere in the '
                           'ROS_PACKAGE_PATH, AMENT_PREFIX_PATH or in any of the directories '
                           'given by the --from-paths option.'

Levi-Armstrong, 2020-02-07:
It is a [bug](https://github.com/ros-infrastructure/rosdep/issues/740) in industrial ci.

closed 2020-02-07 by Levi-Armstrong