ros-infrastructure/rosdep #745: rosdep install suddenly broken on Melodic

icolwell-as, 2020-03-23:
We have a ROS install script that is essentially the same as the [official instructions](http://wiki.ros.org/melodic/Installation/Ubuntu).
Our script randomly started failing CI in the past week or two with the following error:

> sudo: rosdep: command not found

I also confirmed the issue today on a fresh Ubuntu 18.04 install, the official install instructions now fail at the same spot as the CI fail.

What has changed recently? Our script installs `ros-melodic-desktop`, has `rosdep` been removed from `ros-melodic-desktop`?

gavanderhoorn, 2020-03-24:
Nit, but: `rosdep` has never been part of `desktop` or `desktop_full` (nor any of the other metapackages).

See [REP-150: ROS Melodic and Newer Metapackages](https://ros.org/reps/rep-0150.html).

It probably got pulled in via `rosinstall`, `rosinstall_generator` or `catkin_pkg` (or some other (transitive) dependency of other Python tools/packages).

Tiryoh, 2020-03-24:
I've also come across this problem.
Installing `python-rosdep` manually can avoid the "command not found" error.
Seems like the new version(`2.5.5-1bionic.20200303.063336`) of `ros-melodic-rospack` does not requries `python-rosdep` anymore, it only requires `python-rosdep-modules`.
Previously `python-rosdep` was installed just executing `apt install ros-melodic-desktop-full`.
(`python-rosdep` was required by `ros-melodic-rospack` which is required by `ros-melodic-ros-core`.)
Is installing `python-rosdep` manually required from now on to follow the [installation instruction](http://wiki.ros.org/melodic/Installation/Ubuntu)?

yev-d, 2020-03-25:
I have the same issue under different circumstances.
I installed ros-melodic through conda-forge. 

`conda config --add channels conda-forge`
`conda config --set channel_priority strict`
`conda create --name ros --channel conda-forge ros-core ros-actionlib ros-dynamic-reconfigure ros-roslib python=3.6`

however when trying to initialize `rosdep` it, rightfully denies the permissions to /etc/ros/ in the root:
`ERROR: cannot create /etc/ros/rosdep/sources.list.d:`
`	[Errno 13] Permission denied: '/etc/ros'`
And when writing `sudo rosdep init` it give same error: `sudo: rosdep: command not found`

I don't know where else to ask this, but how can I change rosdep to stay within the conda environment folders, so I don't have to sudo it.

_Ubuntu 18.04.4 LTS, 
conda-forge / packages / ros-roslib 1.14.6_

icolwell-as, 2020-03-25:
Thanks @gavanderhoorn for the link to REP-150! 
I've updated my original post to include some more details and show a temporary solution (thanks @Tiryoh) for people coming here with the same problem. I've left my confusion about `rosdep` belonging to `ros-melodic-desktop` to avoid breaking the flow of conversation.

clalancette, 2020-03-25:
@yev-d Your issues are separate; please don't ask unrelated questions on an issue.  Feel free to open up an issue on https://answers.ros.org, and someone can help you there.

@icolwell-as Thanks for the initial report.  And @Tiryoh thanks for the analysis of rospack.  I'll speak with the rospack maintainers and see what the right path forward here is.

dirk-thomas, 2020-03-25:
The right path forward is to update the instructions to install `rosdep` since they are using the command. The removal from `rospack` was intentional and relying on a transitive dependency isn't recommended.

gavanderhoorn, 2020-03-25:
> The removal from `rospack` was intentional and relying on a transitive dependency isn't recommended.

Could be, but this transitive dependency has existed for almost as long as `rosdep` has existed.

This is a rather big breaking change which should have deserved some more attention.

durka, 2020-03-25:
@gavanderhoorn 
> It's also not broken. Just not installed.

I argue this is absolutely broken because the official instructions say to install `ros-melodic-desktop-full` and then run `rosdep` right away. The instructions were not updated.

gavanderhoorn, 2020-03-25:
`rosdep` is not broken. That's what I meant with that comment.

I'm not arguing the instructions are incomplete.

yev-d, 2020-03-25:
> @yev-d Your issues are separate; please don't ask unrelated questions on an issue. Feel free to open up an issue on https://answers.ros.org, and someone can help you there.

@clalancette 
Ok, Sorry about that. For those who stumble on here with similar to my issue, I've posted to [ROS-forum](https://answers.ros.org/question/347533/cant-initialize-rosdep-after-installing-ros-through-conda-forge/) and [conda-forge team](https://github.com/conda-forge/conda-forge.github.io/issues/1014).
Cheers!

tfoote, 2020-03-25:
We should not add rosdep to the runtime variants. It is a workspace management tool and it should be contained in our bootstrap dependency list along with it's peers like rosinstall-generator.

For source installations it should be in the "Prerequisites" section https://wiki.ros.org/Installation/Source

And the binary installations it should be added to "Dependencies for building packages" http://wiki.ros.org/melodic/Installation/Ubuntu This will just require moving the rosdep init/update into that section too.

dirk-thomas, 2020-03-25:
> For source installations it should be in the "Prerequisites" section https://wiki.ros.org/Installation/Source

That page already contains the necessary instructions.

tfoote, 2020-03-25:
Indeed the source installation is already listed correctly. I've fixed the order of the installation instructions in the Melodic and Kinetic install instructions.

flydragon2018, 2020-09-11:
it is not fixed at all.  I met the same problem.  A complete installation guidance without errors is needed

flydragon2018, 2020-09-11:
I spent a few days , still cannot install melodic on ubuntu 18.04 successfully.

flydragon2018, 2020-09-11:
confusion between  python-rosdep and python3-rosdep.  diffrent errors happened.

caioaamaral, 2020-11-02:
> Indeed the source installation is already listed correctly. I've fixed the order of the installation instructions in the Melodic and Kinetic install instructions.

Just notice that `ros:kinetic-ros-core` image are still coming with rosdep installed, maybe is there a missing update?

the same does not happens when using melodic's and noetic's core images.

nuclearsandwich, 2020-11-14:
> confusion between python-rosdep and python3-rosdep. diffrent errors happened.

If you experience other issues using rosdep please either post a new question on https://answers.ros.org or open a new issue.

nuclearsandwich, 2020-11-14:
> Just notice that `ros:kinetic-ros-core` image are still coming with rosdep installed, maybe is there a missing update?

It's likely that the change removing the dependency on the rosdep command line tool has not been backported to Kinetic.

nuclearsandwich, 2020-11-14:
Since the installation instructions (as mentioned here https://github.com/ros-infrastructure/rosdep/issues/745#issuecomment-604065699) have been updated I am closing this issue.

closed 2020-11-14 by nuclearsandwich