ros-infrastructure/rosdep #754: Rosdep 0.19.0 hard-coded path is broken

acarrillo, 2020-04-07:
I have a build system which expects `rosdep` to be invokable via `/usr/bin/rosdep` due to a system-wide installation in my build containers. This assumption appears to have been broken by `0.19.0`:

```bash
bash -c ". /opt/ros/kinetic/setup.bash && python /usr/bin/rosdep update"
python: can't open file '/usr/bin/rosdep': [Errno 2] No such file or directory
```

I noticed a different project with the same issue being described, but on Windows (and using Python3.7); given the timestamp on that issue it may be describing the same event: https://github.com/ros-tooling/setup-ros/issues/99

I'm able to fix this by pinning back `rosdep<0.19.0`.

acarrillo, 2020-04-07:
For my system installation, looks like I can find `rosdep` at `/usr/local/bin/rosdep` now...

thomas-moulard, 2020-04-07:
Looking into it too, this seems to be breaking `setup-ros`

cottsay, 2020-04-07:
I am not seeing a difference in behavior in Fedora packaging. The executable was installed to `/usr/bin/rosdep` as expected.

That said, the change that broke your workflow is almost certainly related to #656.

rotu, 2020-04-07:
Yup. Installing to /usr/bin seems to have been the bug, and /usr/local/bin is actually the correct behavior. I recommend closing this issue.

thomas-moulard, 2020-04-07:
> I am not seeing a difference in behavior in Fedora packaging. The executable was installed to `/usr/bin/rosdep` as expected.
> 
> That said, the change that broke your workflow is almost certainly related to #656.

Thanks Scott, this was indeed the issue!

nuclearsandwich, 2020-04-07:
Echoing what @cottsay and @rotu have said, the switch to using the console_scripts entry points is a smoking gun. I'm not sure why the install path behavior is different between the two mechanisms but I would be inclined to let pip's defaults stand and close the issue as @rotu suggests. Thoughts?

acarrillo, 2020-04-07:
I'm okay with that, I think.I wouldn't want `rosdep` to have to maintain backwards-compatibility due to previous bugs; this isn't the win32api, after all ;)

Thanks for responding, everyone! Good job making the change to `console_scripts`; it is certainly difficult to deal with backwards compatibility in the case where a package was previously doing something incorrect :)

closed 2020-04-07 by nuclearsandwich