ros-infrastructure/rosdep #756: Please add POP! OS to the list of Rosdep sources

lootr5858, 2020-05-01:
Below is the error message:
`ERROR: Rosdep experienced an error: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rhel', 'qnx', 'osx', 'tizen', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'centos', 'manjaro', 'arch', 'alpine']
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.18.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 426, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 435, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 644, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 498, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python3/dist-packages/rosdep2/__init__.py", line 90, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 55, in register_platforms
    register_elementary(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 82, in register_elementary
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python3/dist-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 698, in get_name
    self.detect_os()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 668, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rhel', 'qnx', 'osx', 'tizen', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'centos', 'manjaro', 'arch', 'alpine']`

Please add POP! OS into the list thank you

tfoote, 2020-05-01:
OS detection is implemented in rospkg. 

Docs here: http://docs.ros.org/independent/api/rospkg/html/developers_guide.html#developing-new-osdetectors

You'll find some examples by looking through older PRs: https://github.com/ros-infrastructure/rospkg/pulls?q=is%3Apr+is%3Aclosed

And since pop is an ubuntu derivative you'll probably want to add an alias here: https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/debian.py

This is not a supported platform so we're not likely to spend time working on this but are happy to review pull requests to add support.

lootr5858, 2020-05-04:
May i check if pop os (ubuntu) 20.04 is ready please?

mikaelarguedas, 2020-05-07:
PRs to rospkg and rosdep to add Pop OS! support have been opened:
https://github.com/ros-infrastructure/rospkg/pull/199
https://github.com/ros-infrastructure/rosdep/pull/757

Reminder that in the meantime you should be able to pass `--os ubuntu:focal` or set the environment variable `ROS_OS_OVERRIDE=ubuntu:focal`to use rosdep on Pop! 20.04

christophebedard, 2021-06-26:
> PRs to rospkg and rosdep to add Pop OS! support have been opened:
> [ros-infrastructure/rospkg#199](https://github.com/ros-infrastructure/rospkg/pull/199)
> #757

These PRs have been merged so this issue can be closed.

closed 2021-07-19 by nuclearsandwich