ros-infrastructure/rosdep #759: python3-rosdep-modules and python-rosdep-modules not side-by-side installable with version 0.19.0-1

sloretz, 2020-05-13:
It looks like `python(3)-rosdep-modules` incorrectly depend on the non-"modules" packages on Bionic, and because of that they're not side-by-side installable.

From http://packages.ros.org/ros/ubuntu/dists/bionic/main/binary-amd64/Packages

```
Package: python3-rosdep-modules
Priority: optional
Section: python
Installed-Size: 310
Maintainer: Tully Foote, Ken Conley <tfoote@osrfoundation.org>
Architecture: all
Version: 0.19.0-1
Replaces: python3-rosdep (<< 0.18.0)
Depends: python3-catkin-pkg, python3-rosdistro, python3-rospkg, python3-yaml, python3:any (>= 3.4~), ca-certificates, python3-rospkg-modules (>= 1.1.10), python3-catkin-pkg-modules (>= 0.4.0), python3-rosdistro-modules (>= 0.7.5), sudo
Conflicts: python3-rosdep (<< 0.18.0), python3-rosdep2
Filename: pool/main/p/python3-rosdep-modules/python3-rosdep-modules_0.19.0-1_all.deb
SHA512: f08abea3c1f8efb4bad10b027b4f0129c0bd56c8b70e49e994226f5bebf45bec6a64ea230fa2a810b67341ae74b61ba930e11e0db30b062ef8dddabf0112e8c0
Size: 50552
SHA256: a29a4f8215e6bdc2b44a029beb31e3719a50b288e84a756f478436e7c8b0fd52
SHA1: 431c37ca31925b04bb1d65081bddf045be6d6607
MD5sum: cdab4a33c901c7548ba9c75949a0f185
Description: rosdep package manager abstraction tool for ROS
 Command-line tool for installing system dependencies on a variety of platforms.
```

The problem is `python3-catkin-pkg, python3-rosdistro, python3-rospkg` are present in `Depends` when they should not be. Same problem for the Python 2 package

```
Package: python-rosdep-modules
Priority: optional
Section: python
Installed-Size: 311
Maintainer: Tully Foote, Ken Conley <tfoote@osrfoundation.org>
Architecture: all
Version: 0.19.0-1
Replaces: python-rosdep (<< 0.18.0)
Depends: python (<< 2.8), python (>= 2.7), python-catkin-pkg, python-rosdistro, python-rospkg, python-yaml, python:any (>= 2.6.6-7~), ca-certificates, python-rospkg-modules (>= 1.1.10), python-catkin-pkg-modules (>= 0.4.0), python-rosdistro-modules (>= 0.7.5), sudo
Conflicts: python-rosdep (<< 0.18.0), python-rosdep2
Filename: pool/main/p/python-rosdep-modules/python-rosdep-modules_0.19.0-1_all.deb
SHA512: 2b5931d127d0aa8a8d83274cfd124b67a5b65bd0a31c5f8e4e6f5e10ddd3cc65904a0ae6a63b27c86d60cea85006902de92c23b8b047e171f06787090b685bbd
Size: 50548
SHA256: e3f7fb5bab52dbe15475c1ba88d505db9d1519a809f4f1d3e655430f40d515c2
SHA1: 13419cf7820ea4c7efa9a28060d16ca3244a6a9a
MD5sum: 2bd786f9ce7c4708333159d0b79605cd
Description: rosdep package manager abstraction tool for ROS
 Command-line tool for installing system dependencies on a variety of platforms.
```

The `Depends line incorrectly has `python-catkin-pkg, python-rosdistro, python-rospkg`

Same issue in Focal http://packages.ros.org/ros/ubuntu/dists/focal/main/binary-amd64/Packages, though here it doesn't matter since there's no Python 2 modules package to conflict with.

```
Package: python3-rosdep-modules
Priority: optional
Section: python
Installed-Size: 310
Maintainer: Tully Foote, Ken Conley <tfoote@osrfoundation.org>
Architecture: all
Version: 0.19.0-1
Replaces: python3-rosdep (<< 0.18.0)
Depends: python3-catkin-pkg, python3-rosdistro, python3-rospkg, python3-yaml, python3:any (>= 3.4~), ca-certificates, python3-rospkg-modules (>= 1.1.10), python3-catkin-pkg-modules (>= 0.4.0), python3-rosdistro-modules (>= 0.7.5), sudo
Conflicts: python3-rosdep (<< 0.18.0), python3-rosdep2
Filename: pool/main/p/python3-rosdep-modules/python3-rosdep-modules_0.19.0-1_all.deb
SHA512: f08abea3c1f8efb4bad10b027b4f0129c0bd56c8b70e49e994226f5bebf45bec6a64ea230fa2a810b67341ae74b61ba930e11e0db30b062ef8dddabf0112e8c0
Size: 50552
SHA256: a29a4f8215e6bdc2b44a029beb31e3719a50b288e84a756f478436e7c8b0fd52
SHA1: 431c37ca31925b04bb1d65081bddf045be6d6607
MD5sum: cdab4a33c901c7548ba9c75949a0f185
Description: rosdep package manager abstraction tool for ROS
 Command-line tool for installing system dependencies on a variety of platforms.
```

sloretz, 2020-05-13:
@nuclearsandwich anything different about the 0.19.0 release? I'm not sure where to start looking; it seems like `stdeb.cfg` is ok: https://github.com/ros-infrastructure/rosdep/blob/d1784da592310c7ad8e649530200cdf957a87afc/stdeb.cfg#L19-L20

nuclearsandwich, 2020-05-22:
> @nuclearsandwich anything different about the 0.19.0 release? I'm not sure where to start looking; it seems like `stdeb.cfg` is ok:

No actually. The rosdep packages have never been co-installable. #731 and #750 have both gotten us closer but we're still not there yet.

The problem here is the same dependency guessing issue we face with bloom adding a depending on rosdep2 from upstream Debian/Ubuntu. Unfortunately the workaround we use there (release from a distribution that doesn't have upstream packages to guess is harder since our supported platforms all have either python2 or python3 versions of these packages in upstream. I've got a PR started to add config options for stdeb to control the dependency guessing but it won't be ready today.

dirk-thomas, 2020-05-22:
I would guess `dh` is guessing the additional Debian packages based on the dependencies declare in the `setup.py` file. Since we do need those for `pip` they could be conditionally removed when building `.deb` packages since the necessary dependencies are explicitly enumerated in the `stdeb.cfg` file.