ros-infrastructure/rosdep #760: rosdep install --simulate provided interactive commands

ruffsl, 2020-05-22:
Currently, `--simulate` for `rosdep install` generates commands that may be interactive. e.g.:

```
$ rosdep install --simulate --reinstall \
>       --from-paths \
>         src/ros2/demos/demo_nodes_cpp \
>         src/ros2/demos/demo_nodes_py \
>       --ignore-src
#[apt] Installation commands:
  apt-get install ros-dashing-example-interfaces
  apt-get install python3-pytest
  apt-get install ros-dashing-launch-testing
  apt-get install ros-dashing-launch-testing-ament-cmake
  apt-get install ros-dashing-launch-testing-ros
```

I'd suggest these commands include `-y`, or perhaps recommend uses set `DEBIAN_FRONTEND=noninteractive` when using piped output for scripting.

rafa-martin, 2023-12-05:
Is there any update about this topic? I need it for dockerfile with certain packages.

ruffsl, 2024-03-08:
@rafa-martin , not that I'm aware. I think I once needed this to generate lock files that included package dependency version for CI cache optimization, but ended up just working around it by piping around the stdout from a non simulated install:

- https://github.com/ros-planning/navigation2/blob/47374622dee01a27e5f9b8ae08f3d19a15de9b3a/.circleci/config.yml#L100-L113

ruffsl, 2024-03-21:
I think this is working now? e.g. I can use awk to reformat the stdout to get the package list for use with further apt scripting:

```bash
rosdep install --simulate -y \
    --from-paths src \
    --ignore-src \
  | awk '{print $4}'
```

rafa-martin, 2024-03-22:
Yes, I think so.