ros-infrastructure/rosdep #763: Provide python3-rosdep2 package

sloretz, 2020-05-26:
It looks like on Ubuntu Focal, bash will suggest `python3-rosdep2` when a user types the command `rosdep`. This is the name of the upstream Debian ackage, but with it users are unable to install `ros-noetic-*` packages. Maybe we could provide a `python3-rosdep2` package to hide the upstream one?

I'm not sure how to implement this, so I'm opening a ticket to see if someone has an idea.

https://answers.ros.org/question/353082/missing-packages-after-installing-rosdep-based-on-python3-rosdep2-in-noetic/

dirk-thomas, 2020-05-26:
> It looks like on Ubuntu Focal ...

I would expect this to already be the case in older Ubuntu distros. Since the suggestion comes from what `apt` knowing which Debian package contains an executable with a given name.

sloretz, 2020-05-26:
> I  would expect this to already be the case in older Ubuntu distros.

Ah sure enough. I hadn't noticed because my containers didn't have `command-not-found` installed. Once I install that I see the same message in a Bionic container.

```console
$ /usr/lib/command-not-found rosdep

Command 'rosdep' not found, but can be installed with:

apt install python-rosdep2
```

gavanderhoorn, 2020-05-26:
You'll get the notice on every Ubuntu install which has that package installed.

It's been really rather annoying, and I've answered I don't know how many questions by users which have inadvertently followed the advice printed and installed the upstream(ed) packages.

>  * Current idea is to explore adding ROS packages to the command-not-found database. Maybe `ros-environment` or `catkin` could install it?

Is it really a database? I was under the impression it uses something like a reverse `apt` lookup.

sloretz, 2020-05-26:
> Is it really a database? I was under the impression it uses something like a reverse apt lookup.

No idea. I haven't actually looked into it, but it's the direction I'll be checking first.

ReemAlhosainy, 2024-02-27:
i too installed python3-rosdep2 and due to that noetic packages were deleted.. is there any step to revert the change or i will have to install ros-noetics again

sloretz, 2024-02-27:
> i too installed python3-rosdep2 and due to that noetic packages were deleted.. is there any step to revert the change or i will have to install ros-noetics again

You'll have to uninstall that package and reinstall noetic. The following two commands should do it.

```
sudo apt-get remove python3-rosdep2
sudo apt-get install ros-noetic-desktop-full
```

I don't have any plans to try fixing this and it's not really an issue with rosdep, so I'll close this. If someone wants to investigate the source code for the `packagekit-command-not-found` package appears to be here: https://github.com/PackageKit/PackageKit/blob/main/contrib/command-not-found/pk-command-not-found.c

closed 2024-02-27 by sloretz