ros-infrastructure/rosdep #764: I have a problem using rosdep install command

manarhatem, 2020-06-07:
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --as-root pip:no --skip-keys="python-qt-bindings-qwt5 libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev python-imaging" --skip-keys google-mock --skip-keys python-wxtools

this is the command I use; however, I get this error and deprecation warning :

Deprecation: Even though I have python version 3.8

DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support

That is what I get when I run the above command:
executing command [brew install pyqt5]
Warning: pyqt 5.14.2 is already installed and up-to-date
To reinstall 5.14.2, run `brew reinstall pyqt`
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [pyqt5]

I also tried to unlink pyqt5 and link it again but it doesn't work and I also tried to uninstall pyqt5 and install it again separately but also it doesn't work.

please help!

130s, 2020-07-02:
I think troubleshooting discussion like this is encouraged to take place on https://answers.ros.org/questions/

kennydedavion, 2021-06-29:
> I think troubleshooting discussion like this is encouraged to take place on https://answers.ros.org/questions/

please dear sir, never write this again and never tell anyone again.  If you've ever added something to this address yourself, you certainly haven't met, either with answers to your question, or you've been referred. Just like you now.  It's like throwing a hot potato to someone else because they burned me!  Apropo: when an error occurs and you want to fix it right away with the help of familiar people.  Definitely the answer "within an hour" will help.  Instead of yours: I'll ask you a question about answare and I'll be waiting for an answer for days, and when I get to it, it's just a stupid hate, similar to yours.  Personal experience more than once.  Therefore, please inform everyone you have in the area: if I have nothing significant to say, I am silent!  These tips are completely out of the way of friends.

nuclearsandwich, 2021-07-01:
@kennydedavion the [ROS Support guidelines](https://www.ros.org/support/) explicitly state that [ROS Answers](https://answers.ros.org) should be the first place to go for help if you can't find what you're looking for in the ROS wiki and existing documentation and contributors to ROS are welcome to encourage newcomers to adhere to those guidelines.