ros-infrastructure/rosdep #769: rosdep doesn't find packages I expect it to find (boost, rostest, rospy)

rgov, 2020-07-23:
On Ubuntu 18.04 "Bionic" with `ros-melodic-desktop` installed, and following the instructions at http://wiki.ros.org/rosdep (except I installed `python-rosdep2` because `python-rosdep` doesn't exist):

```
$ sudo rosdep init
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Add distro "ardent"
Add distro "bouncy"
Add distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
Add distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/user/.ros/rosdep/sources.cache

$ rosdep install --default-yes --from-paths src --ignore-src
dpkg-query: no packages found matching ros-melodic-rostest
dpkg-query: no packages found matching ros-melodic-rospy
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ds_sensors: No definition of [boost] for OS version [bionic]
ds_asio: No definition of [boost] for OS version [bionic]
```

rgov, 2020-07-23:
Resolved by installing rosdep via pip.

closed 2020-07-23 by rgov

dirk-thomas, 2020-07-23:
> Resolved by installing rosdep via pip.

You should *not* do so as it is explicitly mentioned on the page you referenced.

Also this is a duplicate discussion: #770.