ros-infrastructure/rosdep #770: Rosdep experienced an error: rosdep OS definition for [liborocos-bfl:ubuntu] must be a dictionary, string, or list

rgov, 2020-07-23:
Ubuntu 18.04 "Bionic", ROS Melodic, `python-rosdep2` package:

```
ERROR: Rosdep experienced an error: rosdep OS definition for [liborocos-bfl:ubuntu] must be a dictionary, string, or list: None
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 381, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 389, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 736, in command_db
    inst_key, rule = d.get_rule_for_platform(os_name, os_version, installer_keys, default_key)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 140, in get_rule_for_platform
    raise InvalidData("rosdep OS definition for [%s:%s] must be a dictionary, string, or list: %s"%(self.rosdep_key, os_name, data), origin=self.origin)
InvalidData: rosdep OS definition for [liborocos-bfl:ubuntu] must be a dictionary, string, or list: None
```

clalancette, 2020-07-23:
There are actually two problems here:

First, the liborocos-bfl key isn't available in Bionic.  You can see that it is `null`ed out here: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3210

Second, the version of rosdep that you are using is out-of-date, and so doesn't support the `null` keys properly.  Thus it prints that stack trace.

Since newer versions of rosdep do support null checks, and the package just isn't available in Bionic, I don't think there is anything to do here.  I'll close this out, but feel free to keep commenting or reopen if you think it is in error.

closed 2020-07-23 by clalancette

dirk-thomas, 2020-07-23:
Maybe the problem should be ticked for the upstream Ubuntu package (https://packages.ubuntu.com/bionic/python-rosdep2) which is not able to parse the current rosdep yaml files anymore.

rgov, 2020-07-23:
I agree. This is a bit of a mess because the documentation says to install a package that doesn't exist, and if you install the newer package, it is not new enough...

Another improvement would be that the list should include a minimum version of `rosdep` that can read it, and `rosdep` should not download lists that are too new, but instead either try to get an older list, or instruct you to update.

dirk-thomas, 2020-07-23:
> This is a bit of a mess because the documentation says to install a package that doesn't exist

Can you please point to the documentation you are referring to.

rgov, 2020-07-23:
http://wiki.ros.org/rosdep

dirk-thomas, 2020-07-23:
The mentioned Debian packages `python3-rosdep` / `python-rosdep` do exist. You need to make sure that you have added the ROS apt repo before though (http://wiki.ros.org/noetic/Installation/Ubuntu#Installation.2FUbuntu.2FSources.Setup_your_sources.list). http://wiki.ros.org/UpstreamPackages describes what you ran into and why you shouldn't use those upstream packages.

rgov, 2020-07-23:
(I am using Melodic, just was using Noetic to point out that that wiki was updated recently so is assumed to be the up-to-date information.)

This is super weird because now I have `python-rosdep` showing up but I can see in my .bash_history:

```
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
sudo apt update
...
sudo apt install ros-melodic-desktop python-catkin-tools
```

Those installed successfully. Then when later I went to install `python-rosdep` it told me to install `python-rosdep2`. I would never have known of the 2-suffixed version if it hadn't given me an error.

tfoote, 2020-09-15:
For future reference please don't explicitly install the python-rosdep2 if you're using our repositories. That's the version packaged by upstream and conflicts with the newer versions we provide: http://wiki.ros.org/UpstreamPackages We don't have a good way to disable the suggestion that Ubuntu provides automatically.