ros-infrastructure/rosdep #772: Error using rosdep to install dashing diademata

odellus, 2020-08-01:
I was following the [installation instructions](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Binary/) for Dashing Diademata and I'm coming across this error running rosdep inside `~/ros2_dashing`

```bash
rosdep install --from-paths ros2-linux/share --ignore-src --rosdistro $CHOOSE_ROS_DISTRO -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 osrf_testing_tools_cpp poco_vendor rmw_connext_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers"

ERROR: Rosdep experienced an error: Unable to handle package.xml format version '3', please update catkin_pkg (e.g. on Ubuntu/Debian use: sudo apt-get update && sudo apt-get install --only-upgrade python-catkin-pkg)
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 383, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 436, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 86, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 153, in find_packages_allowing_duplicates
    path_parsed_packages, warnings_lists = zip(*pool.map(parser, data))
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 253, in map
    return self.map_async(func, iterable, chunksize).get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 572, in get
    raise self._value
AssertionError: Unable to handle package.xml format version '3', please update catkin_pkg (e.g. on Ubuntu/Debian use: sudo apt-get update && sudo apt-get install --only-upgrade python-catkin-pkg)
```

Updating `python-catkin-pkg` tells me that it's already the latest version.

What should I do?

clalancette, 2020-08-03:
The `rosdep` version you are using is way out-of-date.  I suspect that you are using the builtin version of rosdep (`python-rosdep2`) rather than the one provided by ROS (`python-rosdep`).  My suggestion is the following:

```
sudo apt-get purge python-rosdep2
```

And then following the instructions from the beginning again.

In the future, please ask these types of questions on https://answers.ros.org.  That way they are all in one place and are searchable.  I'm going to close this out for now.

closed 2020-08-03 by clalancette

odellus, 2020-08-15:
https://answers.ros.org/question/359586/cannot-figure-out-whats-going-on-with-python3-rosdep2/