ros-infrastructure/rosdep #779: rosdep os version searches for Focal Fossa instead of just focal

flynneva, 2020-09-15:
not sure if this is right but it looks like im running into some rosdep issues on Ubuntu 20.04.

`rosdep install` fails for my package on Ubuntu 20.04 because it cant find `camera-info-manager` and a few others.....see [my issue over at that repo](https://github.com/ros-perception/image_common/issues/170).

after running `rosdep resolve camera_info_manager` it looks like maybe the OS version for `Focal Fossa` should be just `focal`?

```
rosdep resolve camera_info_manager

ERROR: No definition of [camera_info_manager] for OS version [Focal Fossa]

No definition of [camera_info_manager] for OS version [Focal Fossa]
	rosdep key : camera_info_manager
	OS name    : ubuntu
	OS version : Focal Fossa
	Data:
_is_ros: true
		debian:
		  buster:
		    apt:
		      packages:
		      - ros-noetic-camera-info-manager
		fedora:
		  '32':
		    yum:
		      packages:
		      - ros-noetic-camera-info-manager
		osx:
		  homebrew:
		    packages:
		    - ros/noetic/image_common
		ubuntu:
		  focal:
		    apt:
		      packages:
		      - ros-noetic-camera-info-manager
```

flynneva, 2020-09-15:
running with the `--os=ubuntu:focal` param seems to fix the issue:
```
rosdep resolve --os=ubuntu:focal camera_info_manager
#apt
ros-noetic-camera-info-manager
```

nuclearsandwich, 2020-11-14:
Can you still reproduce this? I'm not able to do so in my focal environment?

If you can. Would you include
1. The version of rospkg installed
2. The version of rosdep installed
3. The contents of `/etc/os-release` on the system.

Thanks.

flynneva, 2020-11-14:
i cannot reproduce this issue. it looks like this was fixed, or it was an issue with my setup.

closed 2020-11-14 by flynneva

nuclearsandwich, 2020-11-14:
Thanks for taking the time to check again. I don't recall any rospkg changes to OS detection for Ubuntu Focal so my best guess is that a field in the /etc/os-release file which would normally be expected to contain `focal` instead contained `Focal Fossa` but that is pure conjecture.

ksuszka, 2022-02-16:
I've randomly encountered the same issue. After many hours of debugging I found that updating "distro" package to version 1.7.0 causes this issue. Here is single line which shows the issue:
```
docker run -it -e DEBIAN_FRONTEND=noninteractive ubuntu:focal bash -c "apt-get update && apt-get install -y python3-pip python3-rosdep2 && rosdep update && pip3 install distro==1.7.0 && rosdep resolve ament_cmake_clang_format -v 2>&1 | grep matcher"
```

In my case it wasn't updated intentionally, but it did get updated with some other package.

cottsay, 2022-02-16:
Ah, thanks for reporting your findings, @ksuszka!

This is actually a bug in `rospkg`. I opened this ticket to track resolution: ros-infrastructure/rospkg#241