ros-infrastructure/rosdep #780: Please add Arch support for rosdep

specdrake, 2020-10-02:
```rosdep install camera_calibration``` results in : 
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
camera_calibration: No definition of [rostest] for OS [arch]

gstavrinos, 2020-10-05:
Try `yay -S ros-melodic-camera-calibration`

nuclearsandwich, 2020-11-12:
@specdrake rosdep does support Arch Linux. The error being reported, "No definition [rostest] for OS [arch]" indicates that there is no binary package rosdep database or rosdistro which satisfies that dependency on Arch Linux. This is correct as ROS packages are not part of the official Arch Linux repositories nor is there an official ROS repository for Arch Linux to provide those packages.

As @gstavrinos suggests you can install community supported packages from the [AUR](https://aur.archlinux.org) or using rosinstall_generator to bootstrap your own workspace to build from source.

closed 2020-11-12 by nuclearsandwich