ros-infrastructure/rosdep #782: ERROR: Rosdep experienced an error: [Errno 13] Permission denied:

iayushguptaaa, 2020-10-16:
rosdep version: 0.19.0

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 430, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 526, in _package_args_handler
    lookup = _get_default_RosdepLookup(options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 134, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 594, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 552, in load_cached_sources_list
    with open(cache_index, 'r') as f:
IOError: [Errno 13] Permission denied: '/home/bingo/.ros/rosdep/sources.cache/index'

nuclearsandwich, 2020-11-14:
Does the error persist if you run `sudo rosdep fix-permissions`? If `rosdep update` is run with `sudo` after running `sudo rosdep init` it the created cache files in your home directory will be owned by root.

nuclearsandwich, 2020-11-14:
#787 improves the error message here to match other situations where we detect permissions issues.