ros-infrastructure/rosdep #790: Adding support for macOS Big Sur

Chateaubruyant, 2020-12-09:
It'd be great to be able to use ROS on my mac with macOS 11 (Big Sur) installed.
Thanks for your help and support.

```
rosdep version: 0.20.0

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python3.9/site-packages/rosdep2/main.py", line 428, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/local/lib/python3.9/site-packages/rosdep2/main.py", line 437, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/local/lib/python3.9/site-packages/rosdep2/main.py", line 646, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/local/lib/python3.9/site-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/local/lib/python3.9/site-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/local/lib/python3.9/site-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/local/lib/python3.9/site-packages/rosdep2/platforms/debian.py", line 65, in register_platforms
    register_elementary(context)
  File "/usr/local/lib/python3.9/site-packages/rosdep2/platforms/debian.py", line 95, in register_elementary
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/local/lib/python3.9/site-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/local/lib/python3.9/site-packages/rospkg/os_detect.py", line 710, in get_name
    self.detect_os()
  File "/usr/local/lib/python3.9/site-packages/rospkg/os_detect.py", line 672, in detect_os
    self._os_codename = os_detector.get_codename()
  File "/usr/local/lib/python3.9/site-packages/rospkg/os_detect.py", line 385, in get_codename
    return _osx_codename(*ver[0:2])
  File "/usr/local/lib/python3.9/site-packages/rospkg/os_detect.py", line 363, in _osx_codename
    raise OsNotDetected("unrecognized version: %s.%s" % (major, minor))
rospkg.os_detect.OsNotDetected: unrecognized version: 11.0
```

KishoreMayank, 2021-01-06:
Would really appreciate this! Also running into this error:

`ERROR: Rosdep experienced an error: unrecognized version: 11.1
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.20.0

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/main.py", line 428, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/main.py", line 437, in _no_args_handler
    return command_handlers[command](options)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/main.py", line 646, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/platforms/debian.py", line 65, in register_platforms
    register_elementary(context)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/platforms/debian.py", line 95, in register_elementary
    (os_name, os_version) = context.get_os_name_and_version()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rospkg/os_detect.py", line 710, in get_name
    self.detect_os()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rospkg/os_detect.py", line 672, in detect_os
    self._os_codename = os_detector.get_codename()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rospkg/os_detect.py", line 385, in get_codename
    return _osx_codename(*ver[0:2])
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/rospkg/os_detect.py", line 363, in _osx_codename
    raise OsNotDetected("unrecognized version: %s.%s" % (major, minor))
rospkg.os_detect.OsNotDetected: unrecognized version: 11.`

liyaozone, 2021-01-21:
I have the same problem。。。
Have you any ideas?

harshitsankhla, 2021-02-17:
@liyaozone @KishoreMayank @Chateaubruyant
Check this - https://github.com/ros-infrastructure/rospkg/pull/216