ros-infrastructure/rosdep #791: the following packages/stacks could not have their rosdep keys resolved to system dependencies:

oridong, 2020-12-21:
the issue  mirror: https://github.com/ros-planning/moveit/issues/2465

ROS Distro: [Noetic]
OS Version: VMware16.0.0 build-16894299 + Ubuntu 20.04
Source build

when I try to install according to "https://moveit.ros.org/install/source/"
every step succeeds until
this step "rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}"
produces

> rosdep install -y --from-paths src --ignore-src --rosdistro noetic
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> moveit_ros_visualization: Cannot locate rosdep definition for [pkg-config]
> pilz_industrial_motion_planner: Cannot locate rosdep definition for [liborocos-kdl-dev]
> moveit_ros_occupancy_map_monitor: Cannot locate rosdep definition for [eigen]
> moveit_commander: Cannot locate rosdep definition for [python3]
> rviz_visual_tools: Cannot locate rosdep definition for [libqt5x11extras5-dev]
> moveit_core: Cannot locate rosdep definition for [liborocos-kdl-dev]
> moveit_resources_prbt_support: Cannot locate rosdep definition for [eigen]
> geometric_shapes: Cannot locate rosdep definition for [libqhull]
> moveit_ros_planning: Cannot locate rosdep definition for [eigen]
> moveit_setup_assistant: Cannot locate rosdep definition for [yaml-cpp]
> moveit_tutorials: Cannot locate rosdep definition for [eigen]
> moveit_ros_perception: Cannot locate rosdep definition for [opengl]
> moveit_ros_planning_interface: Cannot locate rosdep definition for [python3]
> moveit_ros_manipulation: Cannot locate rosdep definition for [eigen]
> moveit_kinematics: Cannot locate rosdep definition for [liborocos-kdl-dev]

rosdep update passed

> d@ubuntu:~/Robotic/ws_moveit$ rosdep update
> reading in sources list data from /etc/ros/rosdep/sources.list.d
> Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
> Skip end-of-life distro "ardent"
> Skip end-of-life distro "bouncy"
> Skip end-of-life distro "crystal"
> Add distro "dashing"
> Add distro "eloquent"
> Add distro "foxy"
> Skip end-of-life distro "groovy"
> Skip end-of-life distro "hydro"
> Skip end-of-life distro "indigo"
> Skip end-of-life distro "jade"
> Add distro "kinetic"
> Skip end-of-life distro "lunar"
> Add distro "melodic"
> Add distro "noetic"
> Add distro "rolling"
> updated cache in /home/d/.ros/rosdep/sources.cache

cottsay, 2020-12-21:
There should be more lines near the beginning of the output of `rosdep update`:
```
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in ~/.ros/rosdep/sources.cache
```

I think something is up with your `/etc/ros/rosdep/sources.list.d/20-default.list`. You could try removing it, and re-running `sudo rosdep init`.

oridong, 2020-12-22:
> There should be more lines near the beginning of the output of `rosdep update`:

> I think something is up with your `/etc/ros/rosdep/sources.list.d/20-default.list`. You could try removing it, and re-running `sudo rosdep init`.

yes, the true matter is that " sudo rosdep init"fails, although
"wget https://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list" succeeds, and make it convenience to copy 20-default.list to /etc/ros/rosdep/sources.list.d/。
so I naively think I can skip the first "init "step.What else should I do to circumvent the first step, because it always produce 
>ERROR: cannot download default sources list from:
>https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
>Website may be down.

cottsay, 2020-12-22:
For now, you should be able to work around the issue by running:
```
sudo wget -O /etc/ros/rosdep/sources.list.d/20-default.list https://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
rosdep update
```

The error is originating here: https://github.com/ros-infrastructure/rosdep/blob/ac8484eb382302c5ab8f72355cfd0dce4474e542/src/rosdep2/main.py#L580-L588

The thing that's confusing me is that `rosdep update` should halt if there isn't anything under `/etc/ros/rosdep/sources.list.d`, so there must be an empty file there: https://github.com/ros-infrastructure/rosdep/blob/2667916b87c77e9386b93bae93bc843d4c20eab4/src/rosdep2/main.py#L632-L635

The `rosdep init` error your're seeing happens before any files are created, so `rosdep` didn't create any files in there. Did you put an empty file in there manually?

---

Now to get to the bottom of why `rosdep init` is failing to begin with, the download is happening here: https://github.com/ros-infrastructure/rosdep/blob/ac8484eb382302c5ab8f72355cfd0dce4474e542/src/rosdep2/sources_list.py#L339-L342

This looks like a pretty plain `urlopen` operation, so I can't imagine why it's failing when `wget` is working just fine. Both the raw.github.com and raw.githubusercontent.com URLs are working OK for me. Can you tell me about the environment you're running? Is it just plain Ubuntu 20.04 running in a VM?

Also, is it possible the problem was intermittent? I see that there was an incident at Github several hours before this ticket was opened: https://www.githubstatus.com/history
Have you tried `sudo rosdep init` recently?

cottsay, 2020-12-22:
Ah, I just had a though. Are you using a proxy that might be configured in your environment variables? Try this:
```
sudo -E rosdep init
```

oridong, 2020-12-22:
> Ah, I just had a thought. Are you using a proxy that might be configured in your environment variables? Try this:
sorry , I do use proxy  ,if not I cannot visit because of Network regulation.
but just now , I succeed by the following.
>sudo apt-get install ca-cacert
>sudo c_rehash /etc/ssl/certs
>sudo update-ca-certificates
after these, everything goes well

> d@ubuntu:~/Robotic/ws_moveit$ sudo -E rosdep init
> https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
> ERROR: default sources list file already exists:
> 	/etc/ros/rosdep/sources.list.d/20-default.list
> Please delete if you wish to re-initialize
> d@ubuntu:~/Robotic/ws_moveit$ sudo mv /etc/ros/rosdep/sources.list.d/20-default.list /etc/ros/rosdep/sources.list.d/20-default.list-bak
> d@ubuntu:~/Robotic/ws_moveit$ sudo -E rosdep init
> https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
> Wrote /etc/ros/rosdep/sources.list.d/20-default.list
> Recommended: please run
> 
> 	rosdep update
> 
> d@ubuntu:~/Robotic/ws_moveit$ rosdep update
> reading in sources list data from /etc/ros/rosdep/sources.list.d
> Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
> Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
> Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
> Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
> https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
> 1500.0
> Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
> Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
> Skip end-of-life distro "ardent"
> Skip end-of-life distro "bouncy"
> Skip end-of-life distro "crystal"
> Add distro "dashing"
> Add distro "eloquent"
> Add distro "foxy"
> Skip end-of-life distro "groovy"
> Skip end-of-life distro "hydro"
> Skip end-of-life distro "indigo"
> Skip end-of-life distro "jade"
> Add distro "kinetic"
> Skip end-of-life distro "lunar"
> Add distro "melodic"
> Add distro "noetic"
> Add distro "rolling"
> updated cache in /home/d/.ros/rosdep/sources.cache

now the issue is solved.

closed 2020-12-22 by oridong

yeamu, 2023-09-04:
> Ah, I just had a thought. Are you using a proxy that might be configured in your environment variables? Try this:啊，我只是想到了一个想法。您是否使用可能在环境变量中配置的代理？尝试这个：
> 
> ```
> sudo -E rosdep init
> ```
nice thanks

roboticszeng, 2023-12-15:
> Ah, I just had a thought. Are you using a proxy that might be configured in your environment variables? Try this:
> 
> ```
> sudo -E rosdep init
> ```

thanks!

Sologala, 2024-03-19:
> Ah, I just had a thought. Are you using a proxy that might be configured in your environment variables? Try this:
> 
> ```
> sudo -E rosdep init
> ```

Hero!!!

TienVuHong, 2024-05-28:
> Ah, I just had a thought. Are you using a proxy that might be configured in your environment variables? Try this:
> 
> ```
> sudo -E rosdep init
> ```

It work for me, thanks very much