ros-infrastructure/rosdep #794: rosdep is missing entries for Debian Strech on Melodic after rosdep update

romainreignier, 2021-01-07:
In a Debian Stretch Docker container using image `ros:melodic-ros-base-stretch`, calling `rosdep resolve any_pkg_name_here` gives the error message:

```
ERROR: No definition of [catkin] for OS [debian]
```

Steps to reproduce:
- Create a container based on `ros:melodic-ros-base-stretch`
- `rosdep update`
- `rosdep resolve catkin` (or any other package)

```
docker run -it --rm ros:melodic-ros-base-stretch bash -c "rosdep resolve catkin && rosdep update && rosdep resolve catkin"
```

I have noticed this issue on my package CI [here](https://github.com/ixblue/ixblue_ins_stdbin_driver/runs/1661746242?check_suite_focus=true#step:5:54).

When not calling `rosdep update` before, the resolve works but I have noticed that I have to call `rosdep update` first in my CI workflow, [see here](https://github.com/ixblue/ixblue_ins_stdbin_driver/runs/1661721724?check_suite_focus=true#step:5:25).

Note that the problem is not present on Ubuntu or Debian Buster.

clalancette, 2021-01-07:
In early December, we shut off the Debian Stretch builders for Melodic and marked that build end-of-life.  I believe that this means that rosdep doesn't fetch those keys (at least, by default) anymore.  @nuclearsandwich Can you confirm this?  What's the workaround for people who want to use the older versions of Melodic Debian Stretch packages?

gavanderhoorn, 2021-01-07:
This seems to be limited to ROS packages. Other keys do resolve.

>  I believe that this means that rosdep doesn't fetch those keys (at least, by default) anymore. @nuclearsandwich Can you confirm this?

is it perhaps not seeing ROS pkgs in the caches any more?

cottsay, 2021-01-07:
This is how I would expect rosdep to act after ros/rosdistro#27640.

rosdep no longer believes that there are binary packages available for Stretch, which means that the package has to be installed from source, which rosdep doesn't handle.

gavanderhoorn, 2021-01-07:
That's annoying, as the packages are still available. They're just not updated.

clalancette, 2021-01-07:
@cottsay Assuming that the Debian Stretch packages are available in the snapshot repositories, would reverting ros/rosdistro#27640 make it work?  And is that something you'd suggest we do?

cottsay, 2021-01-07:
There are a couple of reasons that I don't think re-adding stretch is a good idea:
1. All melodic package releases will perform unnecessary bloom generation for stretch
2. As the packages and their dependencies in melodic change, the binary packages won't, and rosdep will probably get confused

I think that since the intent here is to make rosdep use of a snapshot of what the supported state was at some given point, the best way to do that is to tell rosdep to target the index at that point as well.

I don't have any experience with [snapshots.ros.org](http://snapshots.ros.org/), but if you can narrow down the state of `rosdistro` when the sync happened that created the target repo there, you can just target that index when you use rosdep. I tested this targeting the commit before stretch was removed from melodic, and it seems to work as expected:
```
$ rosdep update > /dev/null
$ rosdep resolve --rosdistro melodic --os debian:stretch catkin
ERROR: No definition of [catkin] for OS [debian]
$ ROSDISTRO_INDEX_URL=https://raw.githubusercontent.com/ros/rosdistro/f945c4876b2d82147b50965f165613f83abe536e/index-v4.yaml rosdep update > /dev/null
$ rosdep resolve --rosdistro melodic --os debian:stretch catkin
#apt
ros-melodic-catkin
```

My only concern here is that the index itself is parsed in the state targeted by the ref, but the index cache generated by the buildfarm which is referenced by that index isn't version controlled and therefore contains current data. I'm not sure exactly where the cache vs the actual index data is used, but I would guess it is used whenever dependency data is needed, so `rosinstall_generator` might not work right as melodic's dependency structure changes since the snapshot. That said, if you're using `rosinstall_generator` and building from source, you probably want the supported state of melodic anyway, so you should run `rosdep update` without the index URL override.

Note that the state of the rosdep database is unaffected by `ROSDISTRO_INDEX_URL` when running `rosdep update`, so you'll still be getting the most updated rosdep rules either way.

gavanderhoorn, 2021-01-08:
> I think that since the intent here is to make rosdep use of a snapshot of what the supported state was at some given point, the best way to do that is to tell rosdep to target the index at that point as well.

that could work, provided there is actually a snapshot for Stretch.

I've checked `snapshots.ros.org` and it doesn't appear there are any for Melodic.

If I understand @nuclearsandwich correctly in [Will the snapshot repo provide Debian Stretch packages for Melodic?](https://answers.ros.org/question/328801), this is actually as-it-should-be:

> Short answer: without a policy change, not until Melodic goes end-of-support. 
> 
> The snapshot repositories currently serve two functions:
> 
> 1. Supplying "final" archival snapshots of unsupported rosdistros, initially motivated by the [build farm compromise](https://discourse.ros.org/t/security-issue-on-ros-build-farm/9342/8).
> 
> 2. Providing snapshots of each ROS (1) LTS and ROS 2 sync on their primary platform, in all cases so far the Ubuntu LTS release that they target.
> 
> In the first case we tried to archive everything since we were going to be removing the packages from the primary ROS repositories. For the regular snapshots we started with just the Ubuntu LTS release that each LTS rosdistro targets and the recurring sync snapshots scripts actually have the assumption of one rosdistro :: one distro baked into the current snapshot generation scripts. I've dodged the issue for the final snapshots by doing them all by hand which has been fine since you only do a final snapshot once but it'd be too much for every new sync.
> 
> The snapshot repositories are still in "beta" state. They've become somewhat solidified by the fact that we're using some of the same infrastructure for the package archive but that was a decision made in a time sensitive situation rather than one where we had the luxury of looking out at the horizon.

cottsay, 2021-01-08:
> that could work, provided there is actually a snapshot for Stretch.

Alright, I'm not sure this ticket is a good place to hash that out, but I can't find any repositories associated with snapshots, so I don't know where to file a more appropriate ticket.

Hopefully @nuclearsandwich will be able to provide some more information when he's back from vacation on Monday.

clalancette, 2021-03-30:
I know we've been quiet for a while here, but just to follow-up: the snapshot repository, as it stands, only snapshots the current state of the packages and the package metadata.  It does *not* snapshot the current state of rosdistro, nor the other parts of the ROS infrastructure that goes along with that.

If you'd like to keep using Debian Stretch, the current advice is to point to a fork of rosdistro that is pinned to the last version before Debian Stretch was removed.  It was removed from Melodic in https://github.com/ros/rosdistro/commit/087971b184e795d8934b6a9b8e0d12d95ef52359 , which was on December 10th.  The melodic tag that was right before that is https://github.com/ros/rosdistro/tree/melodic/2020-12-07

tfoote, 2021-04-07:
For reference you can do the following to leverage the existing tags of the rosdistro repository for rosdep and rosdistro queries.

```
# Pick you tag
export ROSDISTRO_TAG=melodic/2020-12-07

# adjust the rosdep sources to use the tag
sed -i "s|ros\/rosdistro\/master|ros\/rosdistro\/$ROSDISTRO_TAG|" /etc/ros/rosdep/sources.list.d/20-default.list
# switch rosdistro to use the tag
export ROSDISTRO_INDEX_URL=https://raw.githubusercontent.com/ros/rosdistro/$ROSDISTRO_TAG/index-v4.yaml
```

With this ability I'm going to close this as melodic on stretch is EOL so we're not going to restore the capability here.

I also documented this over here: http://wiki.ros.org/action/edit/SnapshotRepository

closed 2021-04-07 by tfoote

mikaelarguedas, 2021-04-11:
Is the conclusion that the final snapshot will *not* be created for Stretch on Melodic ?

This has been (and still is) a blocker for retiring the Docker images (https://github.com/docker-library/official-images/pull/8405#issuecomment-662167454),  as we would need to (as we did for all other distros e.g Melodic/Artful) create a final docker image using the "final" snapshot before retiring it completely from the dockerhub buildfarm.

---

Also is there an announcement on discourse mentioning stopping building debs for stretch ? I remember asking in multiple places online and offline when this would happen but never saw a notification this was actually done.

clalancette, 2021-04-11:
> Also is there an announcement on discourse mentioning stopping building debs for stretch ? I remember asking in multiple places online and offline when this would happen but never saw a notification this was actually done.

I did mention it here: https://discourse.ros.org/t/preparing-for-melodic-sync-2020-12-01/17681 .  But I guess I never made a formal announcement about it.

mikaelarguedas, 2021-04-11:
> I did mention it here: https://discourse.ros.org/t/preparing-for-melodic-sync-2020-12-01/17681 . But I guess I never made a formal announcement about it.

Thanks for the pointer! I missed that announcement