ros-infrastructure/rosdep #795: Multiple packages found with name "demo_nodes_cpp" and "demo_nodes_py"

bigrobinson, 2021-01-11:
ERROR: Rosdep experienced an error: Multiple packages found with the same name "demo_nodes_cpp":
- demo_nodes_cpp
- demos/demo_nodes_cpp
Multiple packages found with the same name "demo_nodes_py":
- demo_nodes_py
- demos/demo_nodes_py
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.20.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 430, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 485, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python3/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python3/dist-packages/catkin_pkg/packages.py", line 96, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "demo_nodes_cpp":
- demo_nodes_cpp
- demos/demo_nodes_cpp
Multiple packages found with the same name "demo_nodes_py":
- demo_nodes_py
- demos/demo_nodes_py

cottsay, 2021-01-11:
I think the error message is pretty clear - there are multiple packages in your workspace with the same name, which is not supported.
Try removing the duplicate packages and run the command again.

cottsay, 2021-01-11:
Duplicate of #358, #442, and #461.

We should really reconsider the message stating that a bug should be filed when exception is raised.

clalancette, 2021-01-11:
> We should really reconsider the message stating that a bug should be filed when exception is raised.

For what it is worth, I think that when an exception gets raised, it is a good idea to have people file a bug.  At the very least, that means that there is an error path we haven't handled in the code for `rosdep` itself.  There isn't anything we can do about the underlying problem (that does seem to be something wrong in the workspace), but it would be nice to not have stack traces for errors in end-user tools.

bigrobinson, 2021-01-11:
Thank you for the responses and apologies for the trouble. I believe this is a simple Dockerfile misstep. This arose when running dusty_nv's Dockerfile for building a ros2 container for jetson. I simply switched the ROS_PKG argument from "ros_base" to "desktop". After another, closer inspection, I think he was additionally cloning into the demos when building the ros_base variant. I believe it just ended up cloning it twice when I switched to desktop. I have commented out his extra commands to clone and copy demos and that seems to have done the trick.

closed 2021-01-11 by bigrobinson