ros-infrastructure/rosdep #800: rosdep init returns pkg_resources.DistributionNotFound: The 'rosdep==0.20.0' distribution was not found and is required by the application

vitsensei, 2021-02-05:
I just install ROS the first time today (melodic on Ubuntu 18.04). I installed rosdep without any problem using the following command:

`sudo apt install python-rosdep`

Not sure what went wrong during the process but as soon as I run `sudo rosdep init`, I received the following errror:

```
Traceback (most recent call last):

File "/usr/bin/rosdep", line 6, in <module> from pkg_resources import load_entry_point

File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3088, in <module> @_call_aside

File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3072, in _call_aside f(args, *kwargs) File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3101, in _initialize_master_working_set working_set = WorkingSet._build_master() File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 574, in _build_master ws.require(__requires__)

File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 892, in require needed = self.resolve(parse_requirements(requirements))

File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 778, in resolve raise DistributionNotFound(req, requirers) pkg_resources.DistributionNotFound: The 'rosdep==0.20.0' distribution was not found and is required by the application
```

What I'm confused about is how come rosdep is installed using python2 version, my traceback is showing the rosdep command is using version 3.

Any ideas?

dennis39784, 2021-02-11:
I'm hitting the same problem.   After a lot of digging, I found an answer on ros.org that basically says ROS Melodic does not support Python3.  We would have to roll our own from source and not install using standard installation instructions.   https://answers.ros.org/question/348093/using-ros-melodic-with-python-3/

tfoote, 2021-02-11:
Most things in Melodic will support python3, but as mentioned in that thread you need to build from source to make sure to get the correct dependency resolutions. In general you cannot mix a python2 and python3 based system. If you do so you'll be trying to load python3 modules in python2 or vice versa and if the dependencies are only installed on one side you'll get errors like the above. 

I'm going to close this with no response from @vitsensei. There is clearly python3 being added to your path, which means that something is adjusting your PYTHONPATH. If you need more help please ask a detailed question on answers.ros.org including enough information to reproduce your problem including installation instructions and environment setup.

closed 2021-02-11 by tfoote

vitsensei, 2021-02-11:
Yeah. I think the problem was I also developed in Python 3 prior to using ROS. And because of that, I set my machine to use Python 3 by default. I did not realised that ROS would not override my selection and just run using Python 2.