ros-infrastructure/rosdep #801: rosdep install for noetic on macOS asks for GitHub creds for brew tap ros/noetic

harshitsankhla, 2021-02-17:
Trying to use rosdep to install dependencies for packages in my workspace using - 
rosdep install --from-paths src --ignore-src -r -y

and it asks me to input GitHub credentials - 

Continuing to install resolvable dependencies...
==> Tapping ros/noetic
Cloning into '/usr/local/Homebrew/Library/Taps/ros/homebrew-noetic'...
Username for 'https://github.com':

my creds are not working and gives me repository not found. should I register somewhere to have access to this tap or is there some other way to access an open tap ?

HesselM, 2021-02-18:
I had the same issue when installing Foxy. Turned out that the repo which brew wanted to install does not exists. 
Unfortunately I do not recall how I solved it.

tfoote, 2021-02-18:
This doesn't sound like a rosdep bug, but a bug in one of the rosdep rules that the tutorial setups up for you. Please open the ticket there. This tracker is for the source code of rosdep.

closed 2021-02-18 by tfoote

becschwartz, 2025-04-24:
Hello,

What's in your /etc/ros/rosdep/sources.list.d directory?