ros-infrastructure/rosdep #807: missing python-rospkg>=1.3.0 dependency

GuilhermePSC, 2021-04-20:
rosdep version: 0.20.1
python 2.7

I have a setup that uses [this docker image](https://hub.docker.com/layers/osrf/ros/melodic-desktop-full-bionic/images/sha256-c06dcfd97338602211247b347832c70928b30775e365c6e5869e45b89cb58c99?context=explore), which currently contains `python-rosdep 0.20.0` and `python-rospkg 1.2.10`. Then I use a script that updates `python-rosdep` to latest (0.20.1) through apt and then does `rosdep init`, which throws the following:

```
Traceback (most recent call last):
  File "/usr/bin/rosdep", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3088, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3072, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3101, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 576, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 589, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 778, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'rospkg>=1.3.0' distribution was not found and is required by rosdep
```

So, I assume that `python-rosdep` should now be explicitly [dependent](https://github.com/ros-infrastructure/rosdep/blob/30a25a77e053029238eb5db335a2ddde194c51a4/stdeb.cfg#L5) on `python-rospkg >= 1.3.0` so that `python-rospkg` is also upgraded whenever upgrading `python-rosdep`?

cottsay, 2021-04-20:
Thanks for the report. I'm looking into it.

tfoote, 2021-04-20:
You're correct it should have the full dependency. You will need to install the newer version of rospkg as well. Our toolchain doesn't capture the full version dependency into the debian packaging metadata.

We generally recommend that you make sure to do a full `dist-upgrade` not piecemeal upgrades. In the larger ecosystem there are often corrdinated releases that don't get full version locking dependencies declared.