ros-infrastructure/rosdep #813: rosdep --all-versions failed

CodeBeginnerCn, 2021-06-09:
# Question
Hi !
 I am building ros 2 on ubuntu Linux  according to https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html. 
I install rosdep by this commad ``` sudo apt install -y python3-rosdep ```.
When I typed ``` rosdep --all-versions ```, the program experienced an error, No such file or directory: 'gem'.
I don't know what happen to it.
Could anyone help?
Thanks a lot!   
# Terminal Ouput
```
test@ubuntu:~/ros2_galactic$ rosdep --version
0.20.1
test@ubuntu:~/ros2_galactic$ rosdep --all-versions
0.20.1

ERROR: Rosdep experienced an error: [Errno 2] No such file or directory: 'gem'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.20.1

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 389, in _rosdep_main
    installer_version_strings = installer.get_version_strings()
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/gem.py", line 82, in get_version_strings
    gem_version = subprocess.check_output(['gem', '--version']).strip().decode()
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'gem'
```

christophebedard, 2021-06-09:
This is because `gem` isn't installed. If you don't really need `gem`/ruby, you can just disregard the error. If you install `gem` then the error will go away and `rosdep` will output something like this:

```
$ rosdep --all-versions
0.20.1

Versions of installers:
  apt-get 2.0.5
  pip 21.0.1
  setuptools 53.0.0
  gem 3.1.2
  npm b'6.13.4'
```

If wonder why it fails instead of just printing something like:

$ rosdep --all-versions
0.20.1

Versions of installers:
  apt-get 2.0.5
  pip 21.0.1
  setuptools 53.0.0
  gem (not installed)
  npm b'6.13.4'
``` though.

christophebedard, 2021-06-09:
I opened a PR to fix this: https://github.com/ros-infrastructure/rosdep/pull/815

You can try it out with:

```
$ pip3 install -U git+https://github.com/christophebedard/rosdep.git@handle-uninstalled-installers-when-printing-all-versions
```

CodeBeginnerCn, 2021-06-11:
@christophebedard  Thanks!

emersonknapp, 2021-06-11:
I would suggest reopening this issue until it is actually fixed upstream, when the PR is merged. Until that time I'd say this is an open issue.

CodeBeginnerCn, 2021-06-18:
@emersonknapp 
Thanks for advice. I would reopen the issue.
But it's my pleasure if  you could explain why I shoud reopen it? 
Is there any question if I close it after my question is answered?

emersonknapp, 2021-06-18:
> Is there any question if I close it after my question is answered?

GitHub issues are not for questions - issues track work that needs to be done. This is a bug report, and the bug isn't actually fixed until the code is merged onto the `master` branch.

PR #815 (2021-06-25): Print "not installed" for uninstalled installers with --all-versions
Fixes #813

Instead of raising a `FileNotFoundError` error when a given executable isn't found/installed, just print "not installed":

```
$ rosdep --all-versions
0.20.1

Versions of installers:
  apt-get 2.0.4
  pip 20.0.2
  setuptools 45.2.0
  gem not installed
  npm not installed
```

I thought about catching the most generic error that `subprocess.check_output()` can raise, but this might be enough.
closed 2021-06-25 by nuclearsandwich