ros-infrastructure/rosdep #816: Could not install pcl_conversions using rosdep

atinfinity, 2021-06-14:
## Summary

I tried to install packages via rosdep.
But, I could not install `pcl_conversions` using rosdep.

```
$ rosdep install -r -y -i --from-paths src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros2_ouster: Cannot locate rosdep definition for [pcl_conversions]
Continuing to install resolvable dependencies...
```

## System information

- ROS2 => Dashing
- Operating System / Platform => Ubuntu 18.04 64bit

## Steps to reproduce

### Launch Docker Container

```
$ docker run -it --rm ros:dashing bash
```

### Install packages via rosdep

```
$ mkdir -p ros2_ws/src
$ cd ros2_ws/src
$ git clone https://github.com/ros-drivers/ros2_ouster_drivers -b dashing-devel
$ cd ..
$ apt update
$ rosdep update
$ rosdep install -r -y -i --from-paths src
```

But, I could not install `pcl_conversions` using rosdep.

```
$ rosdep install -r -y -i --from-paths src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros2_ouster: Cannot locate rosdep definition for [pcl_conversions]
Continuing to install resolvable dependencies...
```

On the other hands, I could install packages using rosdep on ROS2 Foxy.

## Related issue

- <https://github.com/ros-infrastructure/rosdep/issues/747>

christophebedard, 2021-06-14:
Short answer: Dashing has been marked as end-of-life (https://github.com/ros/rosdistro/pull/29845), so you should use Foxy or Galactic instead. See http://docs.ros.org/en/rolling/Releases.html

However, you can still proceed if you want. By default `rosdep` won't consider EOL distros (https://github.com/ros-infrastructure/rosdep/pull/647), but if you run `rosdep --include-eol-distros` and re-run `rosdep install ...`, it will work.

atinfinity, 2021-06-15:
@christophebedard Thank you for your comment. 

> Dashing has been marked as end-of-life (ros/rosdistro#29845), so you should use Foxy or Galactic instead. 

I understood the reason. And, it was helpful for me.

> if you run rosdep update --include-eol-distros and re-run rosdep install ..., it will work.

I tried this workaround. It work well. Thanks.

closed 2021-06-15 by atinfinity