ros-infrastructure/rosdep #817: The read operation timed out in "rosdep init"

RobespierreXu, 2021-06-16:
I see the following error on rosdep init in ROS Galactic:

> 
> ~$ sudo rosdep init
> 
> 
> ERROR: Rosdep experienced an error: The read operation timed out
> Please go to the rosdep page [1] and file a bug report with the stack trace below.
> [1] : http://www.ros.org/wiki/rosdep
> 
> rosdep version: 0.20.1
> 
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 144, in rosdep_main
>     exit_code = _rosdep_main(args)
>   File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 428, in _rosdep_main
>     return _no_args_handler(command, parser, options, args)
>   File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 437, in _no_args_handler
>     return command_handlers[command](options)
>   File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 581, in command_init
>     data = download_default_sources_list()
>   File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 340, in download_default_sources_list
>     f = urlopen(url, timeout=DOWNLOAD_TIMEOUT)
>   File "/usr/lib/python3.8/urllib/request.py", line 222, in urlopen
>     return opener.open(url, data, timeout)
>   File "/usr/lib/python3.8/urllib/request.py", line 525, in open
>     response = self._open(req, data)
>   File "/usr/lib/python3.8/urllib/request.py", line 542, in _open
>     result = self._call_chain(self.handle_open, protocol, protocol +
>   File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain
>     result = func(*args)
>   File "/usr/lib/python3.8/urllib/request.py", line 1393, in https_open
>     return self.do_open(http.client.HTTPSConnection, req,
>   File "/usr/lib/python3.8/urllib/request.py", line 1354, in do_open
>     r = h.getresponse()
>   File "/usr/lib/python3.8/http/client.py", line 1347, in getresponse
>     response.begin()
>   File "/usr/lib/python3.8/http/client.py", line 307, in begin
>     version, status, reason = self._read_status()
>   File "/usr/lib/python3.8/http/client.py", line 268, in _read_status
>     line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
>   File "/usr/lib/python3.8/socket.py", line 669, in readinto
>     return self._sock.recv_into(b)
>   File "/usr/lib/python3.8/ssl.py", line 1241, in recv_into
>     return self.read(nbytes, buffer)
>   File "/usr/lib/python3.8/ssl.py", line 1099, in read
>     return self._sslobj.read(len, buffer)
> socket.timeout: The read operation timed out
>

cottsay, 2021-06-16:
Are you using a proxy? If so, you need to invoke sudo in such a way that it passes proxy information to the invoked process.

RobespierreXu, 2021-06-16:
Yes, I change my proxy settings and now it works!
Thank you!!.

closed 2021-06-16 by RobespierreXu