ros-infrastructure/rosdep #818: Cannot locate rosdep definition for [image_transport]

apartridge, 2021-06-20:
Hello,

I'm observing this error message when executing 
`rosdep update && rosdep install --from-paths src --ignore-src -r -y`

(as part of CI test pipeline in https://github.com/zivid/zivid-ros)

```
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Add distro "galactic"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zivid_camera: Cannot locate rosdep definition for [image_transport]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

This happens when using the docker image `ros:kinetic-ros-base-xenial`, however it works when using docker image `ros:melodic-ros-base-bionic`. The relevant package.xml is here: https://github.com/zivid/zivid-ros/blob/master/zivid_camera/package.xml.

This last worked on May 8, but started failing sometime between May and May 18, and has subsequently failed. [This can be seen in our CI pipeline summary.](https://dev.azure.com/zivid-devops/zivid-ros/_build?definitionId=1&_a=summary&repositoryFilter=1&branchFilter=2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2%2C2)

I can manually work around this rosdep error by installing the package `ros-kinetic-image-transport` via apt instead. I'm a little lost as to how to debug ths issue further. This may be the wrong place to post this issue.

christophebedard, 2021-06-20:
Kinetic was marked as end-of-life on May 14th (https://github.com/ros/rosdistro/pull/29529), and you can see that `rosdep update` skips it:

> Skip end-of-life distro "kinetic"

You need to run `rosdep update --include-eol-distros` to get `rosdep` to include EOL distros, see https://github.com/ros-infrastructure/rosdep/pull/647.

apartridge, 2021-06-21:
Thanks for your help @christophebedard, I missed that! :smiley:  Closing this issue.

closed 2021-06-21 by apartridge