ros-infrastructure/rosdep #819: rosdep does not register succesful installation of custom dependencies through pip

Bart-van-Ingen, 2021-06-24:
Hi!

I want to use rosdep to install all dependencies for my ros workspace, to keep it as simple as possible. this includes a python package that I have made myself. I have been able to find the process of doing this as can be seen on my [ROS Answers question](https://answers.ros.org/question/380740/best-practice-installing-custom-dependencies-for-ros2/).

Together with gvdhoorn, I have come to the conclusion that rosdep is unable to successfully register the installation of the custom python project through pip.

The error that is indicate is:

``` bash
Installing collected packages: repo
Successfully installed repo-0.1
ERROR: the following rosdeps failed to install
pip: Failed to detect successful installation of [git+https://github.com/user/repo.git@master]
```

I changed the ```pip_detect()``` function in ```/usr/lib/python3/dist-packages/rosdep2/platforms/pip.py ``` by adding more prints to get more insight , as shown bellow, with the added prints indicated with ``` # <-----------``` 

``` python
def pip_detect(pkgs, exec_fn=None):
    print('pkgs =' + str(pkgs)) # <-----------
    """
    Given a list of package, return the list of installed packages.

    :param exec_fn: function to execute Popen and read stdout (for testing)
    """
    pip_cmd = get_pip_command()
    if not pip_cmd:
        return []

    fallback_to_pip_show = False
    print('exec_fn =' + str(exec_fn)) # <-----------
    if exec_fn is None:
        exec_fn = read_stdout
        fallback_to_pip_show = True
    pkg_list = exec_fn(pip_cmd + ['freeze']).split('\n')
    print('pkg_list = ' + str(pkg_list)) # <-----------

    ret_list = []
    for pkg in pkg_list:
        pkg_row = pkg.split('==')
        if pkg_row[0] in pkgs:
            ret_list.append(pkg_row[0])
    print('ret_list =' + str(ret_list)) # <-----------
    print('fallback_to_pip_show =' + str(fallback_to_pip_show)) # <-----------

    # Try to detect with the return code of `pip show`.
    # This can show the existance of things like `argparse` which
    # otherwise do not show up.
    # See:
    #   https://github.com/pypa/pip/issues/1570#issuecomment-71111030
    if fallback_to_pip_show:
        for pkg in [p for p in pkgs if p not in ret_list]:
            # does not see retcode but stdout for old pip to check if installed
            proc = subprocess.Popen(
                pip_cmd + ['show', pkg],
                stdout=subprocess.PIPE,
                stderr=subprocess.STDOUT
            )
            output, _ = proc.communicate()
            output = output.strip()
            if proc.returncode == 0 and output:
                # `pip show` detected it, add it to the list.
                ret_list.append(pkg)
    print('ret_list fallback =' + str(ret_list))
    return ret_list
```

From this i concluded that for the custom package 

```
pkgs =['git+https://github.com/user/repo.git@master#egg=myCustomPkg']
```
while pip only indicates ```'myCustomPkg==0.1'``` in ```pkg_list```

This shows that rosdep is looking for the wrong key.

gavanderhoorn, 2021-06-25:
A possible solution would be to change `pip_detect(..)` to check for `#egg=` in the "package name" (which really seems to be a URI), split it and only keep the part after `#egg=`.

tfoote, 2021-06-25:
An enhancement to make this work would be appreciated. Though I think that we would likely want to prevent the usage of these rules in the public rosdep database as there would be little to no audit trail like that of published packages.