ros-infrastructure/rosdep #820: feature: install only dependencies of a specific type

Aposhian, 2021-06-25:
Many package managers such as [npm](https://docs.npmjs.com/specifying-dependencies-and-devdependencies-in-a-package-json-file) and [cargo](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#development-dependencies) allow for selectively installing a specific class of dependencies, such as `dependencies` and `devDependencies`.

package.xml files conveniently provide multiple tags to classify what dependencies are needed for:
- `<build_depend>`
- `<exec_depend>`
- `<test_depend>`

However, rosdep doesn't support a way to only install `build_depend` dependencies, or only `exec_depend` and `test_depend` dependencies. This is very frustrating for scenarios where you only want to build, or only want to execute.

For example, in CI, I want a build step to install all build dependencies and compile binaries. I transfer those binaries over to a test step where I want to install exec and test dependencies so I can run tests, and then I transfer the binaries over to a final binary deployment that only installs exec dependencies.

Yes, this is _technically_ possible by manually specifying packages to install, or by using `--skip-keys`, but that defeats the purpose of having a package.xml file with these nice, self-documenting tags.

Here is what the usage could look like:
```
rosdep install --from-paths <src-dir> --build-depend
```
or
```
rosdep install --from-paths <src-dir> --exec-depend --test-depend
```

And the default behavior could be to install everything if none of these flags are specified. `<depend>` tags would also get installed no matter what. So adding these flags would not break backward compatibility.

I would love to work on a PR for this.

christophebedard, 2021-06-25:
There's an open PR for that: https://github.com/ros-infrastructure/rosdep/pull/789

Aposhian, 2021-06-25:
Thanks, didn't find that in my searching.

closed 2021-06-25 by Aposhian