ros-infrastructure/rosdep #824: Problem with rosdep

enhatem, 2021-06-30:
Hello, 

I have a problem with rosdep. I'm somewhat of a beginner when it comes to ROS. 
I recently started using ROS2 Galactic, however, I have a problem whenever I try to do anything related to rosdep.
For example, if I run the following line of code in my terminal:

`sudo rosdep init`

I get the following error:
'
[Traceback (most recent call last):
  File "/usr/bin/rosdep", line 33, in <module>
    sys.exit(load_entry_point('rosdep==0.21.0', 'console_scripts', 'rosdep')())
  File "/usr/bin/rosdep", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/rosdep2/__init__.py", line 45, in <module>
    from .lookup import RosdepDefinition, RosdepView, RosdepLookup, \
  File "/usr/lib/python3/dist-packages/rosdep2/lookup.py", line 44, in <module>
    from .sources_list import SourcesListLoader
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 50, in <module>
    from .gbpdistro_support import get_gbprepo_as_rosdep_data, download_gbpdistro_as_rosdep_data
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 20, in <module>
    from .platforms.redhat import YUM_INSTALLER
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/redhat.py", line 34, in <module>
    from rospkg.os_detect import (
ImportError: cannot import name 'OS_ALMALINUX' from 'rospkg.os_detect' (/usr/local/lib/python3.8/dist-packages/rospkg-1.2.8-py3.8.egg/rospkg/os_detect.py)](url)
'

Is there something I am doing wrong?

Thank you

cottsay, 2021-06-30:
I believe you're using an outdated version of rospkg. Try updating to the latest version, 1.3.0.

enhatem, 2021-06-30:
> I believe you're using an outdated version of rospkg. Try updating to the latest version, 1.3.0.

I already have the latest version. When I try  to update it, I get the following:
`
python3-rospkg is already the newest version (1.3.0-100).
python3-rospkg set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
`

cottsay, 2021-06-30:
The backtrace you posted shows rospkg-1.2.8. It seems you have multiple versions installed.

enhatem, 2021-06-30:
> The backtrace you posted shows rospkg-1.2.8. It seems you have multiple versions installed.

Thank you so much! You were right. I uninstalled rospkg and purged then reinstalled and I don't get that error anymore.

closed 2021-06-30 by enhatem