ros-infrastructure/rosdep #826: ROS Neotic : rosdep init error

WeiJuShang1, 2021-08-10:
sudo rosdep init

ERROR: Rosdep experienced an error: The read operation timed out
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 594, in command_init
    data = download_default_sources_list()
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 340, in download_default_sources_list
    f = urlopen(url, timeout=DOWNLOAD_TIMEOUT)
  File "/usr/lib/python3.8/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.8/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/usr/lib/python3.8/urllib/request.py", line 542, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.8/urllib/request.py", line 1397, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "/usr/lib/python3.8/urllib/request.py", line 1358, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.8/http/client.py", line 1344, in getresponse
    response.begin()
  File "/usr/lib/python3.8/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.8/http/client.py", line 268, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/lib/python3.8/socket.py", line 669, in readinto
    return self._sock.recv_into(b)
  File "/usr/lib/python3.8/ssl.py", line 1241, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/lib/python3.8/ssl.py", line 1099, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

yangzion, 2021-12-06:
Same error here.

Shi-jiayu, 2021-12-16:
The same error.

nuclearsandwich, 2022-01-25:
> socket.timeout: The read operation timed out

The bottom of the stacktrace is a network error: `socket.timeout: The read operation timed out`
The default rosdep sources are downloaded from repositories on GitHub. You will need network access to GitHub.com in order to successfully run `sudo rosdep init`.

ZenithalHourlyRate, 2022-07-09:
> rosdep could handle this better and provide a more user-friendly error message rather than a stacktrace.

That could further refer to a Troubleshooting page as suggested in https://github.com/ros2/ros2_documentation/pull/2842